OrcaSlicer/OrcaSlicer #3745: Model Z axis move is inaccurate.

temmiegold, 2024-01-19:
### OrcaSlicer Version

1.9.0

### OS version

Win11 23H2

### Printer

Neptune 4 Pro

### How to reproduce

Hi, When trying to move a model on the Z axis in the prepare section, the displayed Z value is always half of the real model Z value. without the real model Z value it is makes it very difficult to slice a completing part for a failed print (Power loss).

### Actual results

Z value at 0 set model height to the middle

### Expected results

Z value at 0 should set model origin z to bed.

### Project file & Debug log uploads

no log 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

DVSVIDEO, 2024-01-20:
Yes, indeed.
If I move a 20mm high model down to 0mm (the Z height in the input window is 0mm), then exactly half the height of the model is visible.
The explanation is quite simple: a dashed line coordinate appears in the middle of the geometric extent of the model when it is moved.
The numerical values you see are exactly the coordinates of this model origin.
So if Z=0, then it is exactly halfway down the body.

In the screenshot, you can see that the geometric center of the cube is exactly at the home point of the printer, when all fields show 0.00:
[image]

temmiegold, 2024-01-20:
Error or not, the Z coordinate at 0 should be at the base of the model not halfway. you can see Cura and Prusa slicer for reference

DVSVIDEO, 2024-01-21:
So the correct category for the "error" ticket is: Feature request
And the correct title: the Z point of the model should be the lowest point that touches the bed

It makes a big difference what you write. You have not investigated the phenomenon and you have misnamed it.
Think about the amount of work that goes into developing this software (for free)...

closed 2024-04-28 by github-actions[bot]