OrcaSlicer/OrcaSlicer #3748: Saving a prcoess user preset causes OrcaSlicer to freeze and crash

nicktherushfan, 2024-01-19:
### OrcaSlicer Version

1.9.0

### OS version

Win 11

### Printer

Creality K1

### How to reproduce

Have a process user preset currently selected. Change a parameter, then save the preset. Rename the preset to anything else, then click "OK" to save. The slicer will then freeze and requires a force close.

### Actual results

Once slicer is force closed, the process user preset that was attempted to be saved is not present. This only happens when saving to a new preset name, not when overwriting an existing one.

### Expected results

Process user preset should save and be available in the dropdown list.

### Project file & Debug log uploads

[OrcaSlicer Logs, Jan 19 2024.zip](https://github.com/SoftFever/OrcaSlicer/files/13995310/OrcaSlicer.Logs.Jan.19.2024.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

Phardbean, 2024-01-20:
windows 10

same issue and it makes wish I could smash something ..... wth...

I cannot figure out how to save an alternate settings profile because the crap keeps crashing every time I try, softlock.

Phardbean, 2024-01-20:
6 crashes now .. hoping to get lucky ...

nicktherushfan, 2024-01-20:
Both glad and not glad that it isn't only me. Still crashing every time I try a new name on a preset.

dh0dm, 2024-01-26:
Same Problem here - with Arch Linux OS

Phardbean, 2024-01-27:
It seems to be a name “format” issue but, there isn’t any real guidance
other than NO SPACE!

I have more success after I create the “copy” then change the name of that,
erase copy leave @ printer x   And do my naming flavor before the @

jspiros, 2024-02-02:
Encountering the same issue with the current official v1.9.0 AppImage on Debian testing. Attempting to save an existing user process preset as a copy/separate process preset freezes the app.

The freezing does NOT occur if I choose a system process preset first and (modified or not) save it with the new name. At that point it exists in the user process preset list and I can select and edit it and overwrite it fine without issue.

I did not notice any pattern to the names I tried to use, it froze regardless. It seems more related to the flow given that my workaround works fine and can save as any name.

silvia95guy, 2024-02-21:
Still happening to me on version 1.9.1. Super frustrating not being able to save profiles.

GTreaper, 2024-02-25:
Having the same issue!

FryGuyFlies, 2024-02-26:
Yes, the issue is still there in 1.9.1.  I find that if I allow it to create a copy of the user profile without amending anything, it can save it without crashing.  But as soon as I go to trim/amend anything, that's what causes the crash.

As another user said, if you do it straight from a system profile there is no issue, you can amend and append to any extent, so it seems strange that it should have one from a user profile.  I wonder if it has to do with the location of the save file.

So appending = okay, amending = nono.

Surprised we received an update without this fix.  Doesn't seem stable to me.

isaacST08, 2024-02-27:
Same issue on Linux fedora 39.

I can save settings to an already existing custom preset, but the moment I try to save settings to a new preset (just a new name) it freezes and never recovers. Even when not changing any settings and just trying to save a new preset (exact same settings, just a new name) it freezes just the same. The application is not flagged as unresponsive until I try to close the window. When it happens, my fans spin up and one core is pinned at 100% (the load bounces between cores every few seconds but regardless, 1 core is always at 100%) until the process is terminated. On next app start, if there was unsaved setting, the app never fails to recover them but saving still isn't possible.

To me this seems like a classic infinite loop bug but I'm not familiar enough with the code to be able to help any more than a guess.

abandonedbrain, 2024-02-29:
Same issue here, Win 11, Orca Slicer v1.9.1... if it's a character-use limitation issue, I'd be surprised. I'm using only a slight variation on the standard naming convention (which for some reason uses '@' in the name as well). Tried to rename it to something super simple, same issue. So... over a month and no ACK from devs? I don't really want to go back to Cura. :( 

Devs: at least acknowledge this issue for us. Tell us what log files you'd like, etc. and we'll help!

w1ggle, 2024-03-04:
> Encountering the same issue (or at least I think it's the same issue) with the current official v1.9.0 AppImage on Debian testing. Attempting to save an existing user process preset as a copy/separate process preset freezes the app.
> 
> The freezing does NOT occur if I choose a system process preset first and (modified or not) save it with the new name. At that point it exists in the user process preset list and I can select and edit it and overwrite it fine without issue.
> 
> I did not notice any pattern to the names I tried to use, it froze regardless. It seems more related to the UI flow given that my workaround works fine and can save as any name.

This seems to be the only workaround at the moment. Just have to make a copy from a system preset and edit that. If you want to make a new preset, you cannot edit a custom/user one,. Have to make a copy of a system preset again

wccrawford, 2024-03-09:
> This seems to be the only workaround at the moment. Just have to make a copy from a system preset and edit that. If you want to make a new preset, you cannot edit a custom/user one,. Have to make a [...]

This workaround works for me as well.

isaacST08, 2024-03-12:
Basic testing of what previously did not work now works in the new version v2.0.0-beta.
Everyone who had an issue should try it again to see if any problems persist.

closed 2024-06-18 by github-actions[bot]