OrcaSlicer/OrcaSlicer #3751: UI Issue - Download TImelapse

mon5termatt, 2024-01-20:
### OrcaSlicer Version

1.9.0

### OS version

Windows 10 Pro

### Additional system information

    Computer:
      Operating System                                  Microsoft Windows 10 Pro

    Motherboard:
      CPU Type                                          OctalCore AMD Ryzen 7 5800X3D, 4442 MHz (44.5 x 100)
      Motherboard Name                                  Asus ROG Crosshair VIII Extreme 
      Motherboard Chipset                               AMD Bixby X570, AMD K19.2 FCH, AMD K19.2 IMC
      System Memory                                     32663 MB
      BIOS Type                                         AMI (04/19/2023)

    Display:
      Video Adapter                                     NVIDIA GeForce RTX 3090  (24 GB)

### Printer

Bambu P1S

### How to reproduce

https://github.com/SoftFever/OrcaSlicer/assets/43628254/87b11974-ae5f-4c40-b8f9-46d21c82bd0f

### Actual results

Clicking on the Download actually Clicks on Delete.

### Expected results

click the right button.

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Siggy101, 2024-02-08:
I have this exact same error.  It creates a very real risk of inadvertently deleting a file that you want to keep.  
I am running Windows 11 (just upgraded from Windows 10) and it is the same on both OSs.

jctune, 2024-03-12:
Adding that this is still present in 2.0.0-Beta on Mac_arm64.

There's actually a third "hidden" button to the right of "Download".  Clicking the empty space initiates the video download.  
[image]

mon5termatt, 2024-03-12:
I believe it's sectioned to be half and half. Yet the ui shows room for three buttons.

mon5termatt, 2024-06-11:
It's not.

mon5termatt, 2024-07-15:
Monthly check in

Billy-, 2024-07-31:
> It creates a very real risk of inadvertently deleting a file that you want to keep.

I just deleted a timelapse I wanted to download and keep. Very upset 😞 

Is there any word on if this issue will be prioritised?

I'd like to contribute myself but cpp is not my strong suite, and I can't really find any helpful beginner's contributing documentation. If anyone could point me in the right direction, I'd give it a shot. My limited knowledge has me thinking this might be a relatively easy bug to attempt to fix.

mon5termatt, 2024-08-24:
Monthly check in.

szrudi, 2024-10-23:
There are some details in the comments for an other issue that looks like a duplicate of this:
https://github.com/SoftFever/OrcaSlicer/issues/5736

mon5termatt, 2025-01-06:
Another check in..

mon5termatt, 2025-07-08:
BLAH BLAH BLAH KEEP THIS OPEN BLAH BLAH BLAH.

mon5termatt, 2026-11-09:
Not stale as far as I'm aware