OrcaSlicer/OrcaSlicer #3753: Support "Open in Bambu Studio" links from makerworld

hongkongkiwi, 2024-01-20:
On MacOS, there is a simple app which can change the default program that opens on a specific url scheme called "duti". I would like to make it it so links with "bambustudioopen" open in orcaslicer instead.

However, unfortunately OrcaSlicer only understands it's own custom url scheme. Is it possible to add an additional case where which can also understand if it's opened with bambustudioopen url scheme? This way it is possible to change the application to orcaslicer and have it work properly.
 
https://github.com/SoftFever/OrcaSlicer/blob/b30efa727f6c58d19571b81028f87c0659e6ff07/src/slic3r/GUI/GUI_App.cpp#L5884

hongkongkiwi, 2024-01-20:
On MacOS, you can use a tool called duti to change the app that's opened with a url scheme.

```
brew install duti
duti -s com.softfever3d.orca-slicer bambustudioopen
```

However, as orcaslicer does not know the string bambustudioopen it just opens as normal and doesn't do anything. There's no way that I know of to redirect it without having a tool in the middle to pass through the url scheme to another which is a bit silly since we can just make a one line change here to allow both types of url schemes.

DARKFiB3R, 2024-01-21:
This would be nice.

Culain, 2024-02-25:
+1. i can change the app used in firefox to upen the link and orca slicer starts, but no data can be loaded. I would really love to see this implemented.

wiyoon, 2024-03-18:
You need to modify the source code.

src/slic3r/GUI/GUI_App.cpp

```c++
    if (!url.empty() && (boost::starts_with(url, "orcasliceropen://") || boost::starts_with(url, "bambustudioopen://"))) {
        auto input_str_arr = split_str(url.ToStdString(), "orcasliceropen://");
        if (input_str_arr.size() <=1 ) {
            input_str_arr = split_str(url.ToStdString(), "bambustudioopen://");
        }
``` 

You can see the video in my post.

https://blog.naver.com/wiyoon/223386467252

DARKFiB3R, 2024-04-15:
> You need to modify the source code.

Nice :)

An "Open With" dialog would be cool, because I use both Orca and Bambu quite often.

SoCuul, 2024-07-03:
@SoftFever could you please implement the bambustudioopen protocol so models from makerworld can be imported?

t3chguy, 2024-07-11:
So looks like recent versions claim to support this but do lack `bambustudioopen` which is what MakerWorld uses on Apple devices.

[image]
[image]

Indeed on macOS they only register `bambustudioopen` for whatever reason

[image]

Where Orca as expected uses the same consistent url scheme

[image]

AJolly, 2024-09-21:
Yeah it might be easier to work around this with a userscript in the browser instead until Orca adds native support

closed 2024-09-22 by SoftFever

SoftFever, 2024-09-22:
Thank you all!
I have now added support to download MakerWorld directly on Mac. Since macOS doesn't allow applications to register URLs on the fly, I've configured OrcaSlicer to handle the bambustudioopen:// protocol by default on Mac.

Guilouz, 2024-11-08:
@SoftFever I have BambuStudio and OrcaSlicer but Open function with Bambu Studio always open it in Orca. How I can restore it to open it With Bambu Studio ?

SoCuul, 2024-11-11:
> @SoftFever I have BambuStudio and OrcaSlicer but Open function with Bambu Studio always open it in Orca. How I can restore it to open it With Bambu Studio ?

That has to do with the url associations either in your browser or on your system. You can either change it in your system settings or in your browser settings.

duesk, 2024-11-13:
> @SoftFever I have BambuStudio and OrcaSlicer but Open function with Bambu Studio always open it in Orca. How I can restore it to open it With Bambu Studio ?

`brew install duti
duti -s com.bambulab.bambu-studio bambustudioopen`

sfarkhan, 2024-11-29:
> That has to do with the url associations either in your browser or on your system. You can either change it in your system settings or in your browser settings.

I believe there is a misunderstanding. 

There are two things happening here. There is no open URL option on Mac OS like there is on Windows. Setting the preference in Finder doesn't work with web browsers - I tried it with Safari, Chrome, and Firefox. 

Opening a file in Finder works perfectly fine and before the 2.2 update, worked perfectly from a browser but now Orca Slicer overrides. I contacted Apple support to ask if there is a setting for browsers and they said no, that doesn't exist. This request is to have Orca Slicer's functionality to return to the previous where it doesn't override.

PR #6867 (2024-09-22): support downloading from makerworld on Mac by default.
# Description
support downloading from makerworld on Mac by default.

fix #3753