OrcaSlicer/OrcaSlicer #3757: Can not save filamenet user preset

real-artswan, 2024-01-20:
### OrcaSlicer Version

1.9.0

### OS version

Ubuntu 22.04.3 LTS

### Printer

Ghost6

### How to reproduce

1. Go to edit filament settings
2. Press Save
3. Enter new name

### Actual results

App hangs and doesn't response anymore. Profile is not saved

### Expected results

Profile is saved. App works

### Project file & Debug log uploads

[log.zip](https://github.com/SoftFever/OrcaSlicer/files/13998418/log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

real-artswan, 2024-01-20:
What I can do? I tried to rename edit files in the /home/<>/.config/OrcaSlicer/user/2901517671/filament/ but it seems  overrides them

real-artswan, 2024-01-20:
any thoughts?

alsobrsp, 2024-01-23:
Happening for me in v1.9.0 and v2.0.0-dev.

cbaugher, 2024-01-30:
Also confirmed in 1.9.0.  This is seriously annoying.

You can create custom filaments based on existing presets, but it only seems to work for system preset printers and not user preset printers.

bulcke, 2024-02-02:
I can confirm this in 1.9.0

jhstatewide, 2024-02-22:
Happening in Linux on OS 1.9.1

Buster-Grob, 2024-02-25:
This is still happening for me. Win 11, Orca Slicer 1.9.1. Filament type, name, settings don't matter. When I go to save, it crashes the program and doesnt save the profile.

ichibana, 2024-02-29:
so right now i'm just really dialing in my settings for **Voron Generic ABS - Copy** because I can't rename and save it :'(

DJMitch117, 2024-03-06:
renaming filament profile "Bambu PAHT-CF Calibrated" to "Bambu PAHT-CF 0.4mm Calibrated" causes the app to become unresponsive. This is the only filament user profile of mine that does this. Windows 10, 1.9.1.

glizzyglizz69, 2024-03-18:
New to Orca so glad to see I'm not the only one - happened on the last few nightly builds as well as the stable current build. Win 10.

jdart, 2024-04-13:
As a workaround you can click the gear icon in the top right of the filament panel, in the popup go to the custom filaments tab, and create new.

mrbell321, 2024-07-13:
This appears to be resolved by at least v2.1.1

tmenet, 2024-08-17:
I am using v 2.1.1 I still get an immediate crash when I try to export preset bundle. All other options are greyed out.

Beep6581, 2024-10-22:
Run Orca through GDB. Then, when it hangs or crashes, get a stack backtrace in GDB.

Instructions:
http://rawpedia.rawtherapee.com/How_to_write_useful_bug_reports#Step_by_step

closed 2025-01-28 by github-actions[bot]