OrcaSlicer/OrcaSlicer #3758: Selected object hidden in support painting

claustro, 2024-01-20:
### OrcaSlicer Version

1.9.0

### OS version

Windows both portable and installed

### Additional system information

AMD Ryzen 7 4800HS with Radeon Graphics, 2900 Mhz, 8 core, 16 processori logici
Nome SO	Microsoft Windows 11 Pro Versione	10.0.22631 build 22631
Memoria fisica installata (RAM)	16,0 GB
NVIDIA GeForce GTX 1660 Ti with Max-Q Design	OK

### Printer

Custom made

### How to reproduce

Open a new stl file select it and push L or support painting

### Actual results

I can only se an empty gray space, if I slide the section view slider I can see a black section of the model

### Expected results

See the model  during support painting

### Project file & Debug log uploads

[log.zip](https://github.com/SoftFever/OrcaSlicer/files/13998516/log.zip)
[test.zip](https://github.com/SoftFever/OrcaSlicer/files/13998517/test.zip)
[Screenshot 2024-01-20 171143.zip](https://github.com/SoftFever/OrcaSlicer/files/13998533/Screenshot.2024-01-20.171143.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

ViniSug, 2024-01-20:
I'm having the same problem.

CaptainRoot, 2024-01-21:
I'm having the same problem, too.
But only on my Windows 10 computer, my laptop with Windows 11 works fine.

tech4teachers, 2024-01-24:
Yes, we are seeing this on every computer in the school lab now....
Tried rolling the manjaro machines back to 1.8.1 to no avail, I think it is a video driver issue there
but....
Under Win10 the entire build area goes empty/blank when selecting the support painting, but.... if I put my finger on the screen right where I wanted to paint supports in or out  I can wiggle the mouse and then slice my model and everything is correct ... the spot I painted does indeed get supports added (or remove them). This is interesting because in previous iterations of Orca, choosing the painted supports always zoomed out and then I had to zoom back in and tumble around the view to get it where I wanted.  This means that while the new window rendering is not visible the position of the model stays put and scaled exactly how it was when I activated the support painting. Perhaps going back to the older system which defaults to a different view and requires the user to re-orient  in order to do their painting might make sense.  Unfortunately I can't test in Manjaro KDE right now, a recent update broke all my nvidia drivers and a lot more is wonky than just Orca... though the entire build space is blank all the time ... not just  when painting supports. 

I will report back once I have tried LMDE and an XFCE X-windows variant .... which usually run perfectly.
Sorry I don't have a good way to dump driver calls at the moment, I will work on that as well.

pamafe1976, 2024-01-31:
I have the same problem.

KuroObi72, 2024-02-03:
gleiches Problem bei Stützen malen und Mehrfabig malen, nur grauer Bildschirm

tech4teachers, 2024-02-06:
UPDATE::
This is one weird problem.... 
Windows 10 with an AMD Radeon R7 200 series, is missing the support paint in/out view of the model .... it vanishes as stated earlier....
AND now 
KDE PLASMA Manjaro Linux  using an Nvidia GTX mobile GPU  also suffers from this problem .... whole model disappears and the workspace is just a "blank" empty gray.

.... so it's definitely Orca Slicer 1.9.0, because at home under win10 with my Nvidia GTX1650 I have no issues with 1.8.1
... the same KDE Plasma machine .... same metal... was switched to XFCE manjaro and 1.9.0 is completely fine... support painting is rendered perfectly! 

Independent of GPU, but dependent on the window rendering interface of the Operating System .... whether Win10 or Arch seems especially hard to track down..... we could be looking at more than one issue ...and a coincidence not a common source for both OS's and GPU types.

I'd hoped it was a classic PITA with all Nvidia cards and their proprietary driver inflexibility....but the Radeon not working either tells me this is going to be a tricky one to resolve.  I need to investigate the Vulkan interface, someone suggested there was a potential connection at least for explaining Windows inconsistencies. 

I will try more win10 and win11 and even other Linux installs next, and will see if I can round up more hardware as well.

Need folks to REPLY with their specific GPU model and OS version, maybe even GPU driver version.....

pamafe1976, 2024-02-07:
In my case is AMD Radeon HD 7700, driver version 27.20.20913.2000 and Win10

tech4teachers, 2024-02-09:
This will be my last post for  a while:
Rolling back to 1.8.1 Works on the Radeon R7 250, no drivers ... new or old seem to solve the issue for 1.9.0 where the support painting causes the object to disappear completely... but still be there ( invisibly)  so a finger on the screen or a piece of tape on the monitor can still be blindly painted and "work" but that is such a kludgey option I'm just rolling back to 1.8.1 and painting is flawless every time.

Tried all manner of Nvidia drivers under Manjaro, and LMDE.... none worked ... in fact things got WORSE with Nvidia 470 tesla drivers under LMDE and even the starting welcome screen on new installs of ALL Orca versions 1.7.0 to 1.9.0 failed.

Rolled back to Nouveau and everything is fine again.

If only there was a Nouveau equivalent for the R7 under Windows. 

So .... to sum up...
1.8.1 works with R7 250  in Windows 10 
1.9.0 is broken

Nouveau is required on Nvidia legacy hardware for any of the Linux flavors I tried... all versions will break with any proprietary driver ....full stop.

closed 2024-05-17 by github-actions[bot]