OrcaSlicer/OrcaSlicer #3762: Crashing When Slicing STL with Fuzzy Skin

theeqme, 2024-01-20:
### OrcaSlicer Version

1.9.0

### OS version

macOS Monterey 12.7.2

### Printer

Bambu Lan P1S

### How to reproduce

1. Load in STL
2. Enable Fuzzy Skin - Contour
3. Slice

### Actual results

[image]

Orca Slicer immediately crashes at 15%. Cannot slice regardless of what the fuzzy skin settings are.

### Expected results

Should slice as normal and allow for model to print. All other models with the exact same settings slice and print as expected. Tried restarting and reinstalling and issue continues to be present.

### Project file & Debug log uploads

[[WIP ORCA] PY2A 19.5.3mf.zip](https://github.com/SoftFever/OrcaSlicer/files/13999261/WIP.ORCA.PY2A.19.5.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

BillySp33d, 2024-02-03:
i'm experiencing the same problem on the windows verision of 1.9.0

BillySp33d, 2024-02-03:
> i'm experiencing the same problem on the windows verision of 1.9.0

Also another weird thing i just noticed is it works with 6 objects on the plate,t but as soon as i add a 7th one it results in a crash.

closed 2024-05-13 by github-actions[bot]