OrcaSlicer/OrcaSlicer #3766: Bed incorrectly displayed for delta printer

pmarkiewicz, 2024-01-20:
### OrcaSlicer Version

1.9.0

### OS version

Window 10

### Printer

Kossel mini

### How to reproduce

open slicer and select delta printer
This is regression and this happened after upgrade to 1.9.0

### Actual results

see picture

[image]

### Expected results

bed grid and shadow should be in same place

### Project file & Debug log uploads

[Body12.zip](https://github.com/SoftFever/OrcaSlicer/files/13999463/Body12.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

tastechnician, 2024-01-23:
Hi Piotr.
Issue raised in case #3581. Perhaps you wish to transfer your details there.

kind regards,

pmarkiewicz, 2024-01-24:
Close as there is #3581
https://github.com/SoftFever/OrcaSlicer/issues/3581

closed 2024-01-24 by pmarkiewicz