OrcaSlicer/OrcaSlicer #3769: Top layer is being sliced as a bridge

samwiseg0, 2024-01-21:
### OrcaSlicer Version

1.8.1

### OS version

MacOS 13.4.1 (c) (22F770820d)

### Printer

Voron 2.4 300

### How to reproduce

Open attached project file and slice.

What should print as a top layer prints as a bridge layer. To work around the issue I change the indent to be one layer lower or one layer higher. It seems to be related with that layer sharing the same layer as a bridging layer after the infill. AKA first top layer. 

### Actual results

[image]

### Expected results

indented area to print as a top layer. 

### Project file & Debug log uploads

[body123_.8.3mf.zip](https://github.com/SoftFever/OrcaSlicer/files/13999855/body123_.8.3mf.zip)

[orca_logs.tar.gz](https://github.com/SoftFever/OrcaSlicer/files/13999861/orca_logs.tar.gz)

### Checklist of files to include

- [X] Log file
- [X] Project file

TheMeanCanEHdian, 2024-01-31:
I seem to be running into this issue as well. Initially I had the issue with a 0.6mm indent and found that a 1mm indent worked as expected. Upon further testing a 0.9mm indent it just completely fails to do a top surface at all for the indent. 

Below are my 3mf and log files as well as screenshots showcasing the behaviors. 

[3mf files.zip](https://github.com/SoftFever/OrcaSlicer/files/14117303/3mf.files.zip)
[orca_log.zip](https://github.com/SoftFever/OrcaSlicer/files/14117302/orca_log.zip)

### 0.6mm 
Layer 15
[image]

Layer 16
[image]

Layer 17
[image]

### 0.9mm
Layer 15
[image]

Layer 16
[image]

Layer 17
[image]

### 1mm
Layer 14
[image]

Layer 15
[image]

Layer 16
[image]

TheMeanCanEHdian, 2024-08-01:
This is still occurring on OrcaSlicer 2.1.1.

TheMeanCanEHdian, 2025-02-24:
A similar issue is happening with [this model](https://www.printables.com/model/560892-aztec-pattern-debossed-planter-pot-with-drainage-h/files#preview.file.N9bhX) where it continues to bridge over existing bridges rather than doing solid infill, leading to a messy top layer. Print settings was 5 top shell layers on OrcaSlicer 2.1.1.

[image]

Felix14-v2, 2026-10-06:
Unable to reproduce the problem with all of the provided 3MFs:

[image]

@TheMeanCanEHdian your problem is different: you need to tweak the wall threshold value to achieve preferred result:

| 0.6 | 0.9 | 1.0 |
|:-:|:-:|:-:|
| [image] | [image] | [image] |

But in your case, you should try the new setting from 2.5.0: top surface expansion, it combines the entire surface into one area:

[image]

ianalexis, 2026-10-10:
> 30\. But in your case, you should try the new setting from 2.5.0: top surface expansion, it combines the entire surface into one area

This!

closed 2026-10-10 by ianalexis