OrcaSlicer/OrcaSlicer #3770: Independent Top-Bottom & "Sparse" Infill Direction

manferlo81, 2024-01-21:
**Is your feature request related to a problem? Please describe.**
Sometimes (many times in my case) one needs to have more control over infill direction. Say you want to use bicubic infill but it make more sense to have it at 90 degrees of 60 degrees (for some reason). It would be nice to have a way to control the internal (sparse) infill direction to achieve a more robust part while keeping the nice rectilinear 45 degrees (or some other) infill on top and bottom.

**Which printers will be beneficial to this feature**
All of them

**Describe the solution you'd like**
Implement independent "top-bottom infill direction" & "sparse infill direction". I propose the introduction of "Top-Bottom Infill direction" to control the top & bottom infill angle & the solid infill angle, defaulting to the the same value as the already implemented "Infill direction" so it's backward compatible.

**Describe alternatives you've considered**
You could use "by-layer" settings but it leaves the solid infill at an awkward angle compare with the top-bottom layer, it may cause the part to warp, or worse, to twist.

**Additional context**
Add any other context or screenshots about the feature request here.

manferlo81, 2024-02-13:
I also noticed "Adaptive Cubic" and "Support Cubic" infill are generated in a different direction compared to "Cubic" (there is a 45 degree difference), in fact I've been using "Adaptive Cubic" instead of "Cubic" for small parts, as there is no difference other than direction, but for bigger parts "Adaptive Cubic" is not what I need. I think this feature would really raise the level of control over infill direction.

mtc1122, 2024-05-01:
While the feature you are requesting isn't implemented the way you request, you can currently use height range modifiers to achieve this. While there is a mistake in the tooltip, the "Infill direction" under Strength>Advanced does affect all infills so using the height range modifiers is useful here.

manferlo81, 2024-06-02:
@mtc1122, thanks for your suggestion, I tried that but using height modifiers causes the layer affected by the modifier to not match the previous layer, creaing a potential weak spot, magine you (for some reason) want the "sparse infill" at 180deg and the "solid infill" at (a very nice looking) 45deg, see image and notice what I mean,

[image]

I hope this clarifies what I mean, I believe this feature whould be very helpful, specially for structural parts where you need a very strong part (sparse infill control), but you also want it to look good outside (solid infill control).

manferlo81, 2024-06-08:
Apparently this has been implemented in the new 2.1.0 beta version (#5267), maybe by accident as there is no mention of this feature request. I am testing it now, it looks good. Thank you @SoftFever.

manferlo81, 2024-07-01:
Feature has been implemented in version 2.1.0 (#5267), that closes this issue.

closed 2024-07-01 by manferlo81