OrcaSlicer/OrcaSlicer #3778: Request: Mouse Ear - Outer Only

alwaysmohawk, 2024-01-21:
I love mouse ear. The little brims they make are so easy to remove, and very effective. The trouble I've found is when trying to deboss text on the bottom layer. The mouse ears effectively create an inner brim.

I'd love to have a similar option to normal brim, where I can choose inner, outer, inner and outer.

[image]
[image]
[image]

mlewis-everley, 2024-03-19:
I am having this exact issue. I actually manually add brim ears now by adding cylinders to the item to print, making them 0.3mm tall and setting width and length to ~5mm.

I then manually move these cylinders to any sharp corners.

This is pretty time consuming though. Being able to say "only outside" (like regular brims) would be awesome!

mlewis-everley, 2024-03-19:
I am having this exact issue. I actually manually add brim ears now by adding cylinders to the item to print, making them 0.3mm tall and setting width and length to ~5mm.

I then manually move these cylinders to any sharp corners.

This is pretty time consuming though. Being able to say "only outside" (like regular brims) would be awesome!

EDIT: I also just noticed #2807 which might also solve this issue?

BETLOG, 2024-03-22:
+1
2.0.0-beta
[image]

doudar, 2024-04-04:
I have the same issue. It's really a problem with nylon because I require the mouse ears to prevent corners from warping up and the equipment cases I print have embossed text.

doudar, 2024-04-04:
As a work around - you can place a flat primitive on the plate and enable a brim on it. This brim will conform to your parts:
[image]

TrentBartlem, 2024-09-02:
Related to #5669, possibly fixed by enhancement #2807

TrentBartlem, 2024-12-04:
Still an issue

MagicWagon, 2025-03-12:
Still an issue.