OrcaSlicer/OrcaSlicer #3782: Crashing when resave identical profile

geekgarage, 2024-01-21:
### OrcaSlicer Version

1.9.0

### OS version

Windows 11 Pro 23H2

### Additional system information

CPU: Intel i9 9900K@5GHz
GPU: ASUS GeForce RTX 3080 Ti ROG STRIX OC (Watercooled EKWB Full Cover + Backplate)
RAM: 32GB DDR4 3000MHz Corsair Vengance LPX
Motherboard: Asus Z390 TUF Plus Gaming (Wi-Fi/Bluetooth)

### Printer

Bambu Lab A1 Mini

### How to reproduce

1. Open Orca Slicer
2. Click to edit a fillament profile
3. Select an existing custom profile
4. Click save and write a new name, i changed from "Bambu PLA Matte @BBL A1M - 0,4 Calibrated" to "Bambu PLA Matte - 0,4 A1M Calibrated"
5. Click ok to save as user preset

### Actual results

[image]

### Expected results

It should just save the profile under a new name

### Project file & Debug log uploads

Project
[Untitled.zip](https://github.com/SoftFever/OrcaSlicer/files/14003063/Untitled.zip)

Log
[log.zip](https://github.com/SoftFever/OrcaSlicer/files/14003066/log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

spyrule, 2024-01-22:
I'm getting this exact same problem, on a Qidi X-Plus 3 printer.  Doesn't seem to matter what I call the profile either.

geekgarage, 2024-01-22:
> I'm getting this exact same problem, on a Qidi X-Plus 3 printer. Doesn't seem to matter what I call the profile either.
> 
> AMD Ryzen 7 7800 X3D, 32GB Ram, Windows 11 Pro, Geforce RTX 4070Ti

Yep i tried to remove any sort of special character and nothing, it sill crashes

Edzi11, 2024-01-22:
Happens to me as well

nearIT87, 2024-01-30:
i have the same issue..

closed 2024-05-07 by github-actions[bot]