OrcaSlicer/OrcaSlicer #3784: Allow renaming of printers in LAN Mode

SoCuul, 2024-01-22:
**Is your feature request related to a problem? Please describe.**
Currently printers display an auto-generated name while connected via LAN mode. This makes it confusing to differentiate between printers.

**Which printers will be beneficial to this feature**
Bambu Lab X1/P1/A1

**Describe the solution you'd like**
It would be nice if we could set an alias in OrcaSlicer to display before (or replace) the auto generated name of LAN printers. The functionality could resemble the renaming of printers when connected via online mode.

rspeed, 2024-02-19:
Bambu seems unwilling to implement this in their own software. Presumably because it would require a modification to the printer firmware. But a basic mapping on the client end would indeed serve as a suitable bandaid in the meantime.

SoCuul, 2024-05-20:
@DanBao-Bambu is this planned to be implemented? printer-side would be difficult, so just being able to create an alias for the printer locally would be perfect.

Eg. 3DP-01P-123 would appear instead as "My Printer" after changing a local configuration setting. It would show this alias name in all references to the printer's name, just like printers within bambu cloud.

closed 2024-08-27 by github-actions[bot]

nate1mace, 2025-06-24:
Is there still no solution to this? Seems like Orca-side mapping would quick thing to implement.