OrcaSlicer/OrcaSlicer #3799: When selecting Calibration>>Pressure Advance>>PA Line, Model Results with No Lines, Rather Text "PA Test"

krmtech, 2024-01-22:
### OrcaSlicer Version

1.9.0

### OS version

MacOS 14.2

### Additional system information

2023 MacBook Pro M3 18GB RAM

### Printer

BambuLab P1P

### How to reproduce

1. Go to Calibration>>Pressure Advance>>
2. Select PA Line, Model Results with Text "PA Test"
3. No Lines are generated
4. Cannot run PA Line Test

### Actual results

[image]

### Expected results

Should see model of PA Test Lines

### Project file & Debug log uploads

[debug_Mon_Jan_22_05_26_39_24152.log.0.zip](https://github.com/SoftFever/OrcaSlicer/files/14016521/debug_Mon_Jan_22_05_26_39_24152.log.0.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

derDoc81, 2024-01-23:
Just slice ;)

krmtech, 2024-01-23:
Thanks, just document

krmtech, 2024-01-24:
Thanks, this was already answered by @derDoc81 .

This step should be added to the documentation as it differs from all of the other calibration models.

closed 2024-05-01 by github-actions[bot]