OrcaSlicer/OrcaSlicer #3800: Omit pressure advance from gcode when flavor is set to klipper.

Somerandoguy20, 2024-01-23:
**Is your feature request related to a problem? Please describe.**

Yes, Orca sends a pressure advance command when using single hot end mm set up with klipper. 

A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

With multiple extruders and tool changes It will error out on the klipper end as it doesn't know what extruder to apply pressure advance to.

**Which printers will be beneficial to this feature**
E.g. Voron/Klipper based printer/etc...

All printers with multiple extruder steppers for mm printing.

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

An option to omit pressure advance commands from gcode completely. 

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

Only alternatives so far are switching to prusa slicer or omitting the gcode commands in klipper's gcode editor.

Feature would be to have a check box to omit pressure advance for mm single hot end printers.

**Additional context**
Add any other context or screenshots about the feature request here.

This isn't the first request for an option for this, this style multi material printing is very easy to set up but lacking on the orca front!

rufinus, 2024-01-23:
just because it doesnt work on your setup - you want to remove a very usefull feature for everybody else????

Somerandoguy20, 2024-01-23:
> Have you ticked the pressure advance box in the filament settings? Ideally you want PA tuned by filament and set the value there. If you want it disabled, untick that box

Yes I have, I'm not saying disable it for all but give me an option to remove the command from g code or allow me to define what tool it's being applied to.

Somerandoguy20, 2024-01-23:
> Have you ticked the pressure advance box in the filament settings? Ideally you want PA tuned by filament and set the value there. If you want it disabled, untick that box

I have that unchecked but when using wipe tower a pa command to set pa to 0 for filament ramming is still sent. Either the extruder being used needs to be defined in the slicer or removal of the pa command from wipe tower for a bowden style multi extruder set up. With direct drive there is no issue as the steppers are synced and no error presents itself.

> just because it doesnt work on your setup - you want to remove a very usefull feature for everybody else????

I'm not saying to completely remove this feature, I'd like to have the ability to remove pressure advance commands from gcode. This is duplicated when the wipe tower is active and causes klipper to get confused as the extruder pa values should be passed to are not defined. I guess a feature to allow tool naming so when it sends a pa command the extruder is defined. Or a simple check box to remove this command when using wipe tower.

Somerandoguy20, 2024-01-23:
#2235 same issue as me.

Somerandoguy20, 2024-01-25:
Using a printer with direct drive results in zero errors, this is a bug as the pa value command is under defined on a bowden setup with multiple extruders.

closed 2024-05-03 by github-actions[bot]

erwann17, 2025-01-08:
Hello I have the same problem, have you found a workaround?