OrcaSlicer/OrcaSlicer #3801: Limit on the number of retraction that can be performed on a certain length of filament.

PeterXQC, 2024-01-23:
**Is your feature request related to a problem? Please describe.**
The K1/K1 max's extruder has many flaws, one of them being that even on the newest revision (white collar, no clip), the spring-loaded gears would squeeze the filament enough that if repeated retraction is applied on the same section of filament (i.e. small amount of extrusion while many retractions are performed), the filament gets squeezed in one dimension so that it no longer fit the Teflon tube and hot end, causing a jam. It is practically impossible to remove the deformed filament without disassembling the extruder, causing significant downtime.

**Which printers will be beneficial to this feature**
K1 & K1 Max

**Describe the solution you'd like**
In Creality Print, under the printer setting, there are two variables "Maximum Retraction Count" and "Minimum Extrusion Distance Window" that make sure that at most <Maximum Retraction Count> amount of retraction can be performed over <Minimum Extrusion Distance Window> amount of filament. After testing, this solves the above-mentioned problem effectively.

**Describe alternatives you've considered**
Disabling retraction completely is another option, but issues including stringing become significantly more prominent in that case.

**Additional context**
Granted, this issue isn't all that common. We are expected to extrude more than we retract. However, when printing things from AnkerMake's Makeitreal website (3d printed images and city skylines), a large number of jumps are required, causing this issue.
[image]

igiannakas, 2024-01-23:
This option should not be desirable as it introduces unpredictable stopping of retractions. What you can do is tune when retractions happen. Set:
1. Distance threshold in the extruder tab to 2mm - this stops retractions from happening during small travel moves.
2. Enable reduce retraction over infill in the other tab
3. Disable retract on layer change 

these three options should allow you to reduce retractions to only where it’s absolutely critical to happen to avoid stringing.

TinkerTob, 2024-04-23:
Thanks for chiming in, @igiannakas

> This option should not be desirable as it introduces unpredictable stopping of retractions. What you can do is tune when retractions happen.

I get your point, but I disagree. This option is merely a failsafe and you can't have enough of those.

If the parameters you mentioned are tuned sufficiently, it should not make any difference. However, if they're not, "unpredictable stopping of retractions" for short intervals during the print is far more desirable than a jammed extruder. Besides, "unpredictable" doesn't even cut it, since the slicer can (and must) predict when this will occur and can warn the user if it does - leaving him the choice between (slight) quality impairments or settings adjustments.

As @PeterXQC mentioned, these problems occur only for certain models that induce massive amounts of retractions. This is pretty unpredictable, and when it comes up unexpectedly, I would really prefer a bit of stringing in parts of the print to disassembling my extruder and spending hours on fine-tuning my retractions (or disabling them altogether).

BTW, Cura also has this feature.

Boerny41, 2024-06-15:
+1 

It straight up saves prints in exchange for 4 or 5 strings.

closed 2024-09-21 by github-actions[bot]

zimmra, 2025-06-13:
> This option should not be desirable as it introduces unpredictable stopping of retractions. What you can do is tune when retractions happen. Set:

I normally would agree but there are some prints that you really can't tune to avoid this - in my case if I want to print honecomb storage wall with 3 walls on a 0.6mm nozzle a large number of consecutive amounts of small walls and retractions is causing a clog every single time. I've tried adjusting just about every setting regarding walls and retraction and there doesn't seem to be a way around it other than not having 3 walls.

There are a plethora of settings that require a bit of forethought and should only be enabled in niche situations, I am not sure why this would be any different - if a user has advanced settings turned on they should be prepared to understand what settings they are enabling/disabling.

ph0tic, 2025-07-16:
this is definitely desirable. please please do this. i have a print (and other detailed prints) where small inner (or outer) walls are made. currently im trying print someone elses design for a small drone and the screw holes are messing me up as the printer does the infill for one tiny screw section, then for some reason does one for the opposite arm, and so on counter clockwise, the rotation is expected - but the intricacy of the print makes it go over the same piece of filament or near it to the point where the extruder looses grip and ... thats it! 

the dumb articulating dragons? i cant print small sizes of them because of this.

we definitely need this because given the choice between STRINGING or complete PRINT FAIL ... i'll take the stringing. 
I love ORca, its my favorite slicer. please consider this request.