OrcaSlicer/OrcaSlicer #3803: Export Configs missing printer/process/filament selector

mjonuschat, 2024-01-23:
### OrcaSlicer Version

2.0.0-dev (2024-01-22)

### OS version

MacOS 14.3

### Printer

Voron 2.4

### How to reproduce

1. Go To "File > Export Configs...
2. Select "Printer config bundle (.orca_printer)
3. Click on "Ok"

### Actual results

See error "Please select at least one printer or filament."

### Expected results

Option to choose a printer/process/filament, either as a step in the dialog or after pressing "Ok" on a following page/dialog.

### Project file & Debug log uploads

[debug_Mon_Jan_22_21_48_31_11623.log.0.zip](https://github.com/SoftFever/OrcaSlicer/files/14019015/debug_Mon_Jan_22_21_48_31_11623.log.0.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

mjonuschat, 2024-01-23:
This seems to happen for all User Presets that I have, independent of Printer/Process/Filament Type.
If I take a system preset and save it as a new preset it seems to work...

SethPDA, 2024-02-03:
This issue still persists. Re-creating everything based on a system presets is too much hassle. Please fix.

Alattice, 2024-02-03:
I'm facing the same issue on the Windows 10 1.9.0 version. The checkboxes show up sometimes for nozzle presets but not filament.

aicherm, 2024-02-20:
This issue is also present in the AppImage (OrcaSlicer_Linux_V1.9.1) on Ubuntu 23.10

montyx00, 2024-04-04:
Checked in v2.0 stable and the bug still exists

gijsprins93, 2024-04-18:
Still all processes are missing. However they are in the correct folder. When i try to import them, it says they already exist.

closed 2024-07-25 by github-actions[bot]

maxim-kukushkin, 2024-09-22:
Could someone please re-open this issue? It's still happening on 2.2.0 - any custom printer configuration doesn't appear in the dialog window

Mierscheid, 2024-09-23:
The developers are not interested in you using printers other than those intended by them. But there is a very simple solution to the problem that works perfectly for me: use Cura!

shadowofsoul, 2025-01-20:
For those looking for a workaround, in windows at least, the user settings are at users folder: `C:\Users\[user]\AppData\Roaming\OrcaSlicer\user\default`