OrcaSlicer/OrcaSlicer #3804: Ender3 v3 ke Network connect.

makers-on1y, 2024-01-23:
### OrcaSlicer Version

1.9.0

### OS version

Windows 11

### Printer

Ender 3 V3 KE

### How to reproduce

Click Network Connections
Start Search

### Actual results

My Ender 3 V3 KE does not show up when I run the navigation with the network button next to the printer list.

The firmware version of Ender 3 V3 KE is 1.1.0.12 and I ran Root permission.

However, it does not appear in the network navigation list.
Entering IP manually does not connect.

### Expected results

Normal WLAN connectivity and printing

### Project file & Debug log uploads

[debug_Tue_Jan_23_14_53_19_19180.log.zip](https://github.com/SoftFever/OrcaSlicer/files/14019063/debug_Tue_Jan_23_14_53_19_19180.log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

ninja-, 2024-01-23:
it won't work since the http interface from Creality is not octoprint-compatible API. But since it's possible to upload gcode files from the web interface, as well as start a print, it could be considered a feature request

henrivdr01, 2024-01-23:
Hi. The CR-10 SE should also benefit from such improvement.

Jiriteach, 2024-01-27:
Works for me? - did you install Moonraker and Fluidd or Mainsail? - You need to do this after root to connect from Orca and other slicers.

fufurax, 2024-03-14:
> Works for me? - did you install Moonraker and Fluidd or Mainsail? - You need to do this after root to connect from Orca and other slicers.

Can you give more information or links on "how to" installé Moonraker...
Tks

MScAlanRojas, 2024-04-05:
> > Works for me? - did you install Moonraker and Fluidd or Mainsail? - You need to do this after root to connect from Orca and other slicers.
> 
> Can you give more information or links on "how to" installé Moonraker... Tks

Found a tutorial for installing Fluidd/Mainsail in the KE, it's in spanish tho. Once is done connection to Orca is possible according to the tutorial, but I still have yet to try it out. I'm new to 3D printing and the KE is my first machine but I'm enjoying diving into the topic. 
https://www.youtube.com/watch?v=CaG4Swrbd2k

Jaizu, 2024-07-08:
Creality Print 5.x is based on orca slicer, bambu studio and such, I wonder if we will be able to make Orca Slicer to work with Creality Web UI once it becomes open source: https://github.com/CrealityOfficial/CrealityPrint

fil1986, 2024-08-23:
Всем привет.ender3 v3 ke, хочу подключить его по сети в creality print, но по ip адресу который выдал роутер, появляется ошибка, что не удалось найти принтер. В чем может быть проблема? IP адрес принтера даже не пингуется через термина. Пробовал сбрасывать до заводских настроек, не помогает. Другие ip в сети пингуются нормально.

closed 2024-11-30 by github-actions[bot]