OrcaSlicer/OrcaSlicer #3806: OrcaSlicer_Mac_x86_64_V2.0.0-dev.dmg Doesn't run and is damaged

3DThingy, 2024-01-23:
### OrcaSlicer Version

OrcaSlicer_Mac_x86_64_V2.0.0-dev.

### OS version

Mac OS 14.2.1

### Additional system information

8-core intel i7

### Printer

Adventurer 6M Pro

### How to reproduce

Once run it doesn't work

### Actual results

[image]

### Expected results

[image]

### Project file & Debug log uploads

[OrcaSlicer_Mac_x86_64_V2.0.0-dev.dmg](https://github.com/SoftFever/OrcaSlicer/releases/download/nightly-builds/OrcaSlicer_Mac_x86_64_V2.0.0-dev.dmg)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

derDoc81, 2024-01-23:
Same here with MBP 15" 2018 6-Core i7
MacOS 14.2.1

StephInoP, 2024-01-23:
same here with macbookair 2017
macos 12.7.2

alienboyxp, 2024-01-23:
same here macbook pro 2020
macos ventura 13.4.1