OrcaSlicer/OrcaSlicer #3807: In Device tab, the access code has to be re-entered again after Orca tries to connect to a printer that is offline.

Vantage-DS, 2024-01-23:
When a printer is turned off and you try to access it via the device tab, Orca Slicer obviously cannot connect to it, but the access code is then lost and the next time the printer is physically turned on and accessed from the device tab, we have to re-enter the access code, even though it hasn't changed.

**Which printers will be beneficial to this feature**
P1S/P1S - Maybe X1C?

**Describe the solution you'd like**
Orca Slicer doesn't assume the access code has changed and instead just tries to reconnect again next time the printer is turned on.

closed 2024-05-01 by github-actions[bot]

yachal-upson, 2025-01-10:
See #6158

redvelociraptor, 2025-01-20:
Have the same issue on 2.2.0 on MacOS Sonoma with a P1S in LAN mode.