OrcaSlicer/OrcaSlicer #3808: Ironing Inset

Vantage-DS, 2024-01-23:
When using 0.6 Nozzle (untested with 0.4) - Ironing results in jagged edges on narrow surfaces that need de-burring. 

**Which printers will be beneficial to this feature**
P1S/X1C/P1P

**Describe the solution you'd like**
Similar to Cura, an option to inset the ironing distance from the edge.

**Describe alternatives you've considered**
Tested different ironing flow rates, speeds and line spacing, the result is the same - still jagged.

**Additional context**
Example from Cura -
[image]

Eldenroot, 2024-01-24:
+1 good idea"

giveen, 2024-01-26:
https://github.com/Ultimaker/Cura/blob/36f1932dbb1353f581b55f09624eec8b792243bf/plugins/VersionUpgrade/VersionUpgrade462to47/VersionUpgrade462to47.py#L99

Eldenroot, 2024-01-26:
It is a good start, it could help with the implementation.

arobdog, 2024-03-03:
Agree with this idea, see my feature request post here: https://github.com/SoftFever/OrcaSlicer/discussions/2778#discussioncomment-7877100

closed 2024-06-09 by github-actions[bot]

PeteCondliffe, 2024-06-22:
I would love to see this too

dfosantos, 2025-01-07:
https://github.com/SoftFever/OrcaSlicer/pull/7969