OrcaSlicer/OrcaSlicer #3811: Fail to load camera live stram from Babmulab X1 Carbon "Laden Fehlgeschlagen [2] !"

Sykr0, 2024-01-23:
### OrcaSlicer Version

1.9.0

### OS version

Windows 11 Pro, Version: 22H2, Build: 22621.3007

### Additional system information

AMD Ryzen 9 7900X 12-Core Processor               4.70 GHz

### Printer

Bambu Lab X1 Carbon

### How to reproduce

It´s not possible to load the Camera Live Stream.

I´ve early reinstalled the Windows Media Player, reinstall Orca and Bambu Studio.

At Bambu Studio, it works without problems.

At Orca Sliver I get this Fail:

[image]

### Actual results

Fail

### Expected results

solve...

### Project file & Debug log uploads

[tutk_log_avapi_13696.txt](https://github.com/SoftFever/OrcaSlicer/files/14029578/tutk_log_avapi_13696.txt)

### Checklist of files to include

- [X] Log file
- [ ] Project file

RPINerd, 2024-01-25:
Having the same issue here on linux, exact same error (but in english lol) and still working fine on Bambu Studio. If this isn't the right log file let me know!
[tutk_log_avapi_312976.txt](https://github.com/SoftFever/OrcaSlicer/files/14045752/tutk_log_avapi_312976.txt)

giveen, 2024-01-26:
It may end up being more than just the camera.  I cant view timelapse, even going to the nightly which has the latest pulls from Bambu Studio

[image]

closed 2024-05-03 by github-actions[bot]