OrcaSlicer/OrcaSlicer #3812: Adding more than one printer on the same host

garik345, 2024-01-23:
**Is your feature request related to a problem? Please describe.**
I can't add two printers that connected to one host, host has one ip, but have two klipper ports for two printers.

**Which printers will be beneficial to this feature**
Klipper based printer

**Describe the solution you'd like**
Please add future to connect two or more printers on the same ip.

faith-healer, 2024-01-24:
+1 to this, same issue.
adding printer by ip:port will be a solution, as it done in PrusaSlicer.

tkunchick, 2024-03-01:
[image]

This doesn't work for you? That's how I add my rooted K1

closed 2024-06-07 by github-actions[bot]