OrcaSlicer/OrcaSlicer #3820: Elegoo Neptune 4 Pro

mapm1998, 2024-01-24:
**Is your feature request related to a problem? Please describe.**
My printer, as per manufacturer specifications, can print TPU, ASA, and Nylon. However, I can't add those filaments to my printer.

**Which printers will be beneficial to this feature**
Elegoo Neptune 4, 4 Pro, 4 Plus, 4 Max.

**Describe the solution you'd like**
If possible, list all materials on the configuration page (or according to the manufacturer's specifications). However, it should be noted that it may be necessary to make some changes to the filament profile.

**Describe alternatives you've considered**
Elegoo has a custom version of cura, with several materials. Those files are in a different type of format (.fdm_material). If the solution presented above is not possible, another option would be to add compatibility for this file type.

danshoop, 2024-02-16:
This sounds like a configuration issue where the OP just needs to define these since they're not included in the default profiles and not a bug.

mapm1998, 2024-02-16:
> This sounds like a configuration issue where the OP just needs to define these since they're not included in the default profiles and not a bug.

There is a way to change this setting?

b2un0, 2024-03-02:
same for me. how can i add `TPU` to my printer profile?

timo-christ, 2024-03-09:
Is there a way to configure it myself if Elegoo doesn't do it? I found a profile on reddit and tried to add it myself. I added to Elegoo/fillament/ and added the fillament in the Elegoo.json. But when I start Orca it gets removed again. I assume the system presets are synced from the web?

Trying to add the fillament is really frustrating. The UI is not really verbose in why I can't add the profile. For somebody who works on Orca or who is experienced this frustration might be not understandable but for someone who just got into 3D Printing it's nothing but frustration. I somehow hope an Elegoo engineer sees this thread and adds a Generic TPU Profile for the Neptune 4's 

Is there maybe somewhere an explanation how to create a TPU profile based on the Generic PLA profile?

timo-christ, 2024-03-17:
I contacted the Elegoo Support and they provided me new profiles including a TPU profile. I just asked them if I'm allowed to share them here. In the meanwhile if you want them you can also contact the Elegoo Support.
I also highlighted to them that they should make it available as a System profile within the Slicer. 

BTW: I haven't had the time yet to test the profile yet.

timo-christ, 2024-03-23:
The Support allowed me to share the profile with you. You need to copy the profiles into your configuration folder. If you don't use the default profile but you signed in then you have to copy the content out of the default subfolder into your account (number). 
[OrcaSlicer1.9.0-Neptune4series-cfg-2401080025.zip](https://github.com/SoftFever/OrcaSlicer/files/14722799/OrcaSlicer1.9.0-Neptune4series-cfg-2401080025.zip)

closed 2024-06-30 by github-actions[bot]

dockrbyter, 2024-07-28:
Same here
https://github.com/SoftFever/OrcaSlicer/issues/3242

aenterausername, 2024-11-24:
same issue here,