OrcaSlicer/OrcaSlicer #3821: Filament Management -

tkunchick, 2024-01-24:
**Is your feature request related to a problem? Please describe.**
I have several different types of printers and different configurations and the filament settings are not valid across printers.

**Which printers will be beneficial to this feature**
People with different printers and different configurations that need to be able to easily select filaments without selection the wrong ones.

**Describe the solution you'd like**
Would like to add a tab in the filament section for printers it should be available to.

IE I have 2 ender 3s with dual fans, one with a single fan, and a K1max.

Filaments between the two enders with dual part cooling can be shared, but not with the K1 or the single part cooling due to different part cooling, flow rates, nozzle sizes and so on.

Might even suggest selecting all printers by default and allowing the user to unselect the printers a filament profile is not valid for. Allowing for no change to users who would not like / use the feature.

**Describe alternatives you've considered**
Currently adding printer / nozzle size to filament names, but due to user error, still select the wrong one a lot.

**Additional context**
This is more a user error / quality of life issue that would make errors less likely.

anselor, 2024-02-26:
If you add a custom filament type it then generates profiles specific to a printer/nozzle. When you select that printer/nozzle profile only the filaments that apply to that printer/nozzle are shown.

tkunchick, 2024-02-26:
> If you add a custom filament type it then generates profiles specific to a printer/nozzle. When you select that printer/nozzle profile only the filaments that apply to that printer/nozzle are shown.

So you mean create a filament, instead of using one of the defaults, like e3-pla+ while having the ender selected? Then build the filament profile?

is it possible to share them between printers? I have 2 Enders, though one on klipper and one on Octoprint, they share the same filament profile, but not the same gcode.

anselor, 2024-02-26:
Yes. You can create a 'base' filament profile and then generate printer/nozzle specific filament profiles that inherit from the base. 
Anything that's in the base profile is shared by the profiles that inherit from it.

If you open up the JSON file for the base one you can add additional printers to the "compatible_printers" field.
For the printer-specific ones that inherit from it you'd only list the single printer in that field.

I don't think this can be done in the GUI but you can modify the files manually to get what you want.

tkunchick, 2024-02-28:
> Yes. You can create a 'base' filament profile and then generate printer/nozzle specific filament profiles that inherit from the base. Anything that's in the base profile is shared by the profiles [...]

Thanks, it's a round about way to get there, but it does work for filtering the filaments to the printer and nozzle.

I hope they implement an easier way.

You are able to add other printers right in the GUI when creating the custom filament. though since it works based on the printer preset, rather than the customized printers, for me at least the ender3s share the filaments by default since they were made off the system preset printer.

Would still rather see a check box or other mechanism to select the user preset printer, rather than system preset printer. since you could have several enders 3s (or other models that use the system preset printer), but have different hot end configurations and there for need different filament profiles.

tkunchick, 2024-02-28:
And after updating all my filament profiles, when trying to open a saved project with mulitpal color changes, orca slicer just crashes.

Seems fine opening projects with no color changes.

Notes:
Extracting libpas PGM metadata failed.

Crashed Thread:        21

Exception Type:        EXC_BAD_ACCESS (SIGBUS)
Exception Codes:       KERN_PROTECTION_FAILURE at 0x000000017863ffd0
Exception Codes:       0x0000000000000002, 0x000000017863ffd0

Termination Reason:    Namespace SIGNAL, Code 10 Bus error: 10
Terminating Process:   exc handler [96703]

VM Region Info: 0x17863ffd0 is in 0x17863c000-0x178640000;  bytes after start: 16336  bytes before end: 47
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      VM_ALLOCATE                 178524000-1785e8000    [  784K] rw-/rwx SM=PRV  
      GAP OF 0x54000 BYTES
--->  STACK GUARD                 17863c000-178640000    [   16K] ---/rwx SM=NUL  ...for thread 21
      Stack                       178640000-178a48000    [ 4128K] rw-/rwx SM=PRV  thread 21

Error Formulating Crash Report:
Extracting libpas PGM metadata failed.

Thread 21 Crashed:
0   OrcaSlicer                    	       0x103437a14 auto Slic3r::lower_bound_by_predicate<std::__1::__deque_iterator<Slic3r::Preset, Slic3r::Preset*, Slic3r::Preset&, Slic3r::Preset**, long, 16l>, Slic3r::PresetCollection::find_preset_internal(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>> const&)::'lambda'(auto const&)>(auto, auto, Slic3r::PresetCollection::find_preset_internal(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>> const&)::'lambda'(auto const&)) + 0
1   OrcaSlicer                    	       0x10341fef4 Slic3r::PresetCollection::find_preset_internal(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>> const&) + 280
2   OrcaSlicer                    	       0x103417320 Slic3r::PresetCollection::find_preset(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>> const&, bool, bool) + 208
3   OrcaSlicer                    	       0x1034192fc Slic3r::PresetCollection::get_preset_base(Slic3r::Preset const&) const + 308
4   OrcaSlicer                    	       0x10341930c Slic3r::PresetCollection::get_preset_base(Slic3r::Preset const&) const + 324
5   OrcaSlicer                    	       0x10341930c Slic3r::PresetCollection::get_preset_base(Slic3r::Preset const&) const + 324
6   OrcaSlicer                    	       0x10341930c Slic3r::PresetCollection::get_preset_base(Slic3r::Preset const&) const + 324
7   OrcaSlicer                    	       0x10341930c Slic3r::PresetCollection::get_preset_base(Slic3r::Preset const&) const + 324
8   OrcaSlicer                    	       0x10341930c Slic3r::PresetCollection::get_preset_base(Slic3r::Preset const&) const + 324
9   OrcaSlicer                    	       0x10341930c Slic3r::PresetCollection::get_preset_base(Slic3r::Preset const&) const + 324
10  OrcaSlicer                    	       0x10341930c Slic3r::PresetCollection::get_preset_base(Slic3r::Preset const&) const + 324
11  OrcaSlicer                    	       0x10341930c Slic3r::PresetCollection::get_preset_base(Slic3r::Preset const&) const + 324
12  OrcaSlicer                    	       0x10341930c Slic3r::PresetCollection::get_preset_base(Slic3r::Preset const&) const + 324
13  OrcaSlicer                    	       0x10341930c Slic3r::PresetCollection::get_preset_base(Slic3r::Preset const&) const + 324
14  OrcaSlicer                    	       0x10341930c Slic3r::PresetCollection::get_preset_base(Slic3r::Preset const&) const + 324
15  OrcaSlicer                    	       0x10341930c Slic3r::PresetCollection::get_preset_base(Slic3r::Preset const&) const + 324
16  OrcaSlicer                    	       0x10341930c Slic3r::PresetCollection::get_preset_base(Slic3r::Preset const&) const + 324
17  OrcaSlicer                    	       0x10341930c Slic3r::PresetCollection::get_preset_base(Slic3r::Preset const&) const + 324
18  OrcaSlicer                    	       0x10341930c Slic3r::PresetCollection::get_preset_base(Slic3r::Preset const&) const + 324
19  OrcaSlicer                    	       0x10341930c Slic3r::PresetCollection::get_preset_base(Slic3r::Preset const&) const + 324
20  OrcaSlicer                    	       0x10341930c Slic3r::PresetCollection::get_preset_base(Slic3r::Preset const&) const + 324
21  OrcaSlicer                    	       0x10341930c Slic3r::PresetCollection::get_preset_base(Slic3r::Preset const&) const + 324
22  OrcaSlicer                    	       0x10341930c Slic3r::PresetCollection::get_preset_base(Slic3r::Preset const&) const + 324
23  OrcaSlicer                    	       0x10341930c Slic3r::PresetCollection::get_preset_base(Slic3r::Preset const&) const + 324
24  OrcaSlicer                    	       0x10341930c Slic3r::PresetCollection::get_preset_base(Slic3r::Preset const&) const + 324
25  OrcaSlicer                    	       0x10341930c Slic3r::PresetCollection::get_preset_base(Slic3r::Preset const&) const + 324
26  OrcaSlicer                    	       0x10341930c Slic3r::PresetCollection::get_preset_base(Slic3r::Preset const&) const + 324
27  OrcaSlicer                    	       0x10341930c Slic3r::PresetCollection::get_preset_base(Slic3r::Preset const&) const + 324
28  OrcaSlicer                    	       0x10341930c Slic3r::PresetCollection::get_preset_base(Slic3r::Preset const&) const + 324
[... cut ...]
507 OrcaSlicer                    	       0x10341930c Slic3r::PresetCollection::get_preset_base(Slic3r::Preset const&) const + 324
508 OrcaSlicer                    	       0x10341930c Slic3r::PresetCollection::get_preset_base(Slic3r::Preset const&) const + 324
509 OrcaSlicer                    	       0x10341930c Slic3r::PresetCollection::get_preset_base(Slic3r::Preset const&) const + 324
510 OrcaSlicer                    	       0x10341930c Slic3r::PresetCollection::get_preset_base(Slic3r::Preset const&) const + 324

closed 2024-06-08 by github-actions[bot]