OrcaSlicer/OrcaSlicer #3822: Multi-layer bridge settings (Cura type bridge settings) to improve bridges for high speed printing

sleewok, 2024-01-24:
**Is your feature request related to a problem? Please describe.**
Currently we can only specify bridge settings for the actual bridge on a single layer. The next layer on top of the bridge will use whatever print speed you have set for the solid infill (or whatever it is I cannot recall 100%). When printing at high speed & acceleration this can cause lots of issues with the bridge quality. Bridge walls get pushed up and the bridge is more likely to sag, etc.

Previously I used Cura and the software includs advanced settings that allowed for modifying multiple (skin) layers following the initial bridge layer. This has a massive impact on the quality of bridges.

**Describe the solution you'd like**
I would like to see something similar implemented in Orca Slicer. You can find details on the bridge settings here: https://community.ultimaker.com/topic/22195-introducing-the-experimental-bridging-settings/

Here is a snippet of the settings: 
[image]

AndreySemjonov, 2024-01-26:
Just opened same issue https://github.com/SoftFever/OrcaSlicer/issues/3847
I agree additional bridge settings missing.

Buzut, 2024-02-18:
I couldn't agree more! At high speed especially with PETG, the second layer doesn't stick to the actual bridged one and blobs of plastic end up sticking to the nozzle.

Workaround so far is to add modifier ranges with lower speed for the "internal solid infill". Very tedious but the only way around it if you don't wanna slow down the whole print.

The way Cura does it is really good, at the very least, in a simple way, we could have a setting like "Slow down for x layers"

dstulken, 2024-02-24:
This would be fantastic if it could be implemented here!

sleewok, 2024-02-24:
> I couldn't agree more! At high speed especially with PETG, the second layer doesn't stick to the actual bridged one and blobs of plastic end up sticking to the nozzle.
>
> Workaround so far is to add [...]

That's a great workaround for now. I didn't even think about that. It will definitely be a pain if you have a lot of bridges, but would be worth it.  Hopefully this gets implemented in some form.

sleewok, 2024-05-26:
Is there any reason this hasn't been addressed? This is an area severely lacking. Superslicer at least has independent fans speeds for internal/external bridges as well as overhangs.

wbagnall, 2024-06-14:
This setting needs to be added. I am forced to use SuperSlicer for some of my prints because without an above-bridge speed setting, the bridges get destroyed and the print fails. I'd argue this issue should take priority, this setting can make or break bridge quality.

Retx3D, 2024-08-08:
I also vote for this feature request. I have the same problem when printing PETG bridges, I can make very long distance bridges and the 1st layer is near perfect (nearly no sagging) but when printing second layer, printer goes at initial speed and destroy everything.

stanpete78, 2024-08-17:
Please add this. Bridges are so much worse on fast printers.

NotoriousEagle, 2024-09-24:
How is this still not adressed... Can't be too hard to add that feature and it would drastically improve bridge quality. Could just implement a multiple layer bridge hight setting.

jpa00, 2024-09-30:
Bridges also very easily get mangled if the solid infill of the layer above happens to not be aligned with the bridge direction well enough, so multi-layer bridges would really be greatly appreciated. This is another one where a modifier can be used as a workaround, but it quickly gets pretty tedious if you have a bunch of models on the same plate and most have several bridges of varying sizes going in multiple directions at different heights. 😅

sheariley, 2024-10-06:
Another upvote for this feature request. PLEASE ADD THIS SOON!

sleewok, 2024-10-12:
Trying to print petg today and the third layer after the initial bridge rips it apart. The lack of ability to control speeds for two layers after the bridge has a huge impact on print quality. It is unprintable.

pterodaktil02, 2024-11-01:
PLEASE ADD THIS SOON!

sutty5, 2025-01-05:
Feature is required, refer to #3847

igiannakas, 2025-02-01:
Hi all

Some ground work has been done on this PR here: https://github.com/SoftFever/OrcaSlicer/pull/8263

There is of course plenty more to be done, but the "hard work" of identifying and segmenting the regions above bridges is now complete.

igiannakas, 2025-02-01:
**Test results from that PR:**
Left - Old / Right - Two bridge layers enabled
Bridge is significantly better
[image]

closed 2025-02-12 by SoftFever