OrcaSlicer/OrcaSlicer #3825: Supprt wall loops

Alexx40in, 2024-01-25:
Please add the option "Support wall loops" from Bambu Studio or a similar option "With steath around the suooprt" from Prusasliser

[image]

[image]

faith-healer, 2024-02-07:
+1 there, need that

AJolly, 2024-03-14:
Its weird, I can search and it says the settings there, but i cant access it

closed 2024-07-05 by github-actions[bot]

toddanglin, 2024-08-19:
Rather than creating another issue, reviving this issue since it's been asked several times and it's precisely what I came here to say/ask. Can anyone shed light on why this setting is available in Bambu Studio/PrusaSlicer but not in OrcaSlicer?

What's odd, is that in the past OrcaSlicer has behaved similarly (adding support walls), but now I'm unable to control that setting and the generated supports are worse.

digital0785, 2025-02-07:
why is this closed as not planned when it's a thing in everything this is based off and is kinda necessary for quality support interfaces vs them dangling at the edges 

i swear this was in there before until searching for this just now

toddanglin, 2025-02-07:
@digital0785 FWIW, I have since found this setting *is* available in Orca, but only when you're using the following Support settings:

1) Type: Tree (manual OR auto)
2) Style: Tree Slim OR Tree Strong OR Tree Hybrid

When using those Support types, you'll see the setting for Support Wall Loops become available. ¯\_(ツ)_/¯

digital0785, 2025-02-07:
> [@digital0785](https://github.com/digital0785) FWIW, I have since found this setting _is_ available in Orca, but only when you're using the following Support settings:
>
> 1. Type: Tree (manual OR [...]

yes but it's needed for standard supports as well. I've done this in super and prusa slicer ages ago

prusa 2.6
[image]

[image]

vs orca 2.3

[image] 

this specific model it's less of an issue but there is 0% chance it works here .. because it's trying to turn around over thin air .. it made that whole area look like crap on a 15 hour print lol.. heck even just having the option to rotate JUST the first interface layer would resolve the issue to be able to get it to align better with the supports but. using grid kinda works but it'll still have ends that dangle over but it's better then default for sure. 

[image]

supports is the absolute worst part of orca it's super frustrating that it never gets any work and given the other 3 slicers that are off the same backbone done have the issue theres no excuse .. 

and with multi material being a thing for having 0 gap supports kinda need reliable interfaces lol. For the last model switching to tree supports not only adds 5 hours but doesn't really save filament  it's 10g less which is pointless.

digital0785, 2025-02-07:
> Agree. It would be nice to have full control over this support setting under all conditions. Not sure why Orca made the decision to hide it for traditional supports.
> 
> In one of my primary use cases where I need this control, I still need traditional supports, and I've found using "Tree Hybrid" will generate traditional supports where I need them (while giving me access to the Support Wall Loops). Clearly, YMMV.

yea usually that works but on this particular model for whatever reason it uses ~10g more and takes longer LOL this is same settings just hybrid tree vs snug normal lol

[image]

[image]

digital0785, 2025-02-07:
> [@digital0785](https://github.com/digital0785) FWIW, I have since found this setting _is_ available in Orca, but only when you're using the following Support settings:
>
> 1. Type: Tree (manual OR [...]

in an odd twist of fate looks like i've discovered a bug.. this actually works IF and only if you have build plate only off lol also apparently support critical means something completely different to orca then myself because a 90+ overhang isn't "critical" lol

[image]

[image]

[image] 

so i guess the "work around" is hybrid with blockers or manual.. but definitely isn't working proper.. devs PLEASE put some effort into supports/ bridging it's really a huge short coming and everything that gets opened about it gets closed for "not planned" when its quite literally the worst part of the slicer.. I'll gladly take a delay on some new features coming out for proper working anchored bridges / multi layer bridges and proper supports without bugs

PlasmaKirill, 2025-02-19:
+1 really need this!
How to add an outer 1 wall for the supports (support outline)? 
It's in the PrusaSlicer, but I can't find it in OrcaSlicer.

OrcaSlicer (?)
[image]

PrusaSlicer (support_material_with_sheath)
[image]

digital0785, 2025-02-19:
> +1 really need this! How to add an outer 1 wall for the supports (support outline)? It's in the PrusaSlicer, but I can't find it in OrcaSlicer.
> 
> OrcaSlicer (?) [image]
> 
> PrusaSlicer (support_material_with_sheath) [image]

this dev build has it :) they're working on it currently
[image]

listed here https://github.com/SoftFever/OrcaSlicer/pull/8212

PlasmaKirill, 2025-02-20:
How to download this 2.3?

digital0785, 2025-02-20:
> How to download this 2.3?

They mention how to get it in that post