OrcaSlicer/OrcaSlicer #3829: MakerWorld - "Open in BambuStudio" -> Point to OrcaSlicer

DokuKaefer, 2024-01-25:
Hello,

if I adjust the path to OrcaSlicer in the registry under Windows and then click on "Open in BambuStudio" on MakerWorld, OrcaSlicer opens but it does not find any geometry in the loaded file.

Is it possible to adjust this? 😊

[image]

[image]

vordenken, 2024-01-30:
Yes please, it would be awesome to use this feature (on Windows as well as macOS).

wiyoon, 2024-03-18:
You need to modify the source code.

src/slic3r/GUI/GUI_App.cpp

```c++
    if (!url.empty() && (boost::starts_with(url, "orcasliceropen://") || boost::starts_with(url, "bambustudioopen://"))) {
        auto input_str_arr = split_str(url.ToStdString(), "orcasliceropen://");
        if (input_str_arr.size() <=1 ) {
            input_str_arr = split_str(url.ToStdString(), "bambustudioopen://");
        }
``` 

https://blog.naver.com/wiyoon/223386467252

closed 2024-06-24 by github-actions[bot]