OrcaSlicer/OrcaSlicer #3830: Attempting to create a new filament fails due to missing printer or preset

ChristianMP, 2024-01-25:
### OrcaSlicer Version

1.9.0

### OS version

OS: Windows 10

### Printer

Ender 3 S1 Pro

### How to reproduce

1. Click filament gear
2. Add new custom filament
3. Select Anycubic TPU
4. Createbased on current filament dropdown is empty
4.1 Copy current filament preset yields same error.

### Actual results

[image]

### Expected results

Filament is created

### Project file & Debug log uploads

No logs created.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

acidg, 2024-02-23:
I am having the same problem: I want to add a custom filament for my custom printer, but can't do so, because the printer cannot be selected. It works for the Bambu Lab X1 Carbon, which I have in my list of printers. Maybe the problem is caused by the custom printer?

acidg, 2024-02-23:
Workaround: Save a profile with another name and adjust all parameters manually.

JediMasterBob, 2024-03-09:
having the same issue in V1.9.1. I've added a custom printer profile and now I do not get any printers in the dropdown box. eventually the program crashes with this error:

[image]

closed 2024-06-16 by github-actions[bot]

RadioAndrea, 2025-02-25:
Present in 2.3.0-Beta as well

steadmanc, 2025-05-16:
I have the same problem.  Cannot add a custom filament because it keeps saying no printer selected.  It does not show any printers to select when you create the custom filament as seen in the instructional video https://www.youtube.com/watch?v=a4PX-2u6BMM.  Fast forward to 1:31 to see where the printer selection is made.  That does not appear for me.