OrcaSlicer/OrcaSlicer #3831: Unable to see devices tab

HandsomeToast, 2024-01-25:
### OrcaSlicer Version

1.9.0 Official release 

### OS version

Windows 11

### Printer

Elegoo 4 Max

### How to reproduce

This is my first ever Github report so sorry if it is a bit jank.

you open the device tab and you see a white blank screen

(I am currently facing an issue with Orca Slicer where I cannot access the Devices tab on the latest version, as well as previous versions like 1.7.0. Whenever I try to access the tab, all I see is a blank white screen. I would greatly appreciate any assistance in resolving this issue.)

### Actual results

I got a blank white screen

### Expected results

being able to send my Sliced files to my printer over my network from my laptop

### Project file & Debug log uploads

[debug_Tue_Jan_09_09_47_40_10452.log.zip](https://github.com/SoftFever/OrcaSlicer/files/14057416/debug_Tue_Jan_09_09_47_40_10452.log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

Irishman0816, 2024-02-11:
im now all of a sudden having this same issue
[image]

dahenson54, 2024-02-22:
I just updated to 1.9.1, and I am still having the same issue. My screen looks the same as Irishman0816.

jheiselman, 2024-02-23:
Version 1.9.1 on MacOS is showing the same behavior

dhoard, 2024-02-28:
Version 1.9.1 on Ubuntu 24.04 is showing the same behavior.

norris108, 2024-02-29:
I have the same, except it's a red screen!

Macbook Pro, Sonoma 14.2.1, Orca Slicer 1.9.1

[image]

technomooney, 2024-03-03:
Same behavior on windows 10. i was looking at the logs and there is no updates to the debug log at all when i select the device tab. it seems the tab does nothing at all... i see it loads the about:blank page, so i assume there is something wrong with the "url" where it starts loading the page. 

i got that info by right clicking and selecting "print" and it shows the url on the top of the print preview. so it has something to do with the built in browser.

HandsomeToast, 2024-03-04:
> Same behavior on windows 10. i was looking at the logs and there is no updates to the debug log at all when i select the device tab. it seems the tab does nothing at all... i see it loads the [...]

Could you show me where the "print" button is? I am not able to find it :D

technomooney, 2024-03-04:
> Could you show me where the "print" button is? I am not able to find it :D

I am right clicking on the blank screen and the print option shows up.

HandsomeToast, 2024-03-05:
> I am right clicking on the blank screen and the print option shows up.

Ah I see thank you I also have just an 'about:blank' page in the print view and it doesn't show any obvious errors in the logs as well

norris108, 2024-03-06:
> I have the same, except it's a red screen!
>
> Macbook Pro, Sonoma 14.2.1, Orca Slicer 1.9.1 [...]

It seems to happen if I've left Orca Slicer open for a long time

technomooney, 2024-06-20:
This has been fixed for me. I dont know about others but I think this can be closed.

HandsomeToast, 2024-06-20:
Yeah I don't have this issue anymore as well I'm going to close this

norris108, 2024-06-20:
I've not had the issue for a while now

tinsarfal, 2024-07-08:
Version 2.1.1 on an Apple Silicon Mac is showing the same behavior.

tolistim, 2024-07-23:
I am seeing the same result - no connection to the printer and a solid blank screen - in 2.1.1 under Linux (Debian, Fedora, and Ubuntu).  It was working in 2.1.0 BETA.  Using a standard browser and connecting to the printer shows the Fluidd interface properly.

stoggi, 2024-08-02:
Not sure if this fixes your issue, I had the same thing with a white screen on the Device tab. It turned out that I was using octoprint over HTTPS with a self signed certificate. Switching to a proper certificate signed by letsencrypt fixed the issue for me.

ronklob, 2024-09-17:
I also see this issue with 2.1.1(amd) on M1 Mac.  I don't have octoprint, so that is not a fix for me.

pilotniq, 2024-10-20:
I'm a new Octo Slicer user. I get this on 2.2.0-rc on MacOS (ARM) 14.6.1 with a printer running via OctoPrint 1.10.2.

pilotniq, 2024-10-20:
> I'm a new Octo Slicer user. I get this on 2.2.0-rc on MacOS (ARM) 14.6.1 with a printer running via OctoPrint 1.10.2.

In my case, the problem was that I did not understand what to write in the "Device UI" field in the physical printer dialog. Once I wrote the hostname of my Octopi host ("octopi.local"), I did get the web page login in the Device tab.

It would have been helpful to get an error like "URL xxx could not be loaded" or something similar.

knowtheory, 2024-10-22:
I'm also new to Orca Slicer and have 2.2.0-rc on an ARM Mac.  I have successfully sent prints to one of my mk4s (with mmu no less!) via Prusa Link auth (both HTTP digest or API key work), but despite that the Device page remains a blank white screen.

I'm not sure where to even start debugging this.  Is there a debug console for OrcaSlicer?

ronklob, 2024-12-13:
It appears that this is now fixed for me in v2.2.0.  Thanks!

rcompton, 2024-12-19:
This is happening to me still on v2.2.0

M4 Mac Mini

ottopic, 2025-01-04:
Same problem for me, has no one found a solution?

tolistim, 2025-01-04:
I'm also still seeing this with a fresh build of nightly, but I sometimes receive an additional message from Webkit:

WebKit encountered an internal error. This is a WebKit bug.

Unfortunately there is no further information and no useful traceback/stack info in a strace run.

rcompton, 2025-01-06:
For my flashforge I had to add extra parameters when connecting the printer:

http://xxx.xxx.x.xxx:8080/stream_simple.html 

I'm sure other printers have their own special urls. After I upgraded to Klipper I had to change this again. Nothing wrong with orca.

NEOLabs-software, 2025-04-11:
i am on latest version and screen goes to about:blank after i switch out of it

[image]

NEOLabs-software, 2025-04-14:
I fixed it by editing the code
For the people experiencing a normal blank screen, try connecting though a browser 
For me, I found a if statement for the webview that checks if I switched away
Removed it

alidorauxier, 2025-05-06:
A quick fix just in case it helps, I'm currently having the same behaviour in Ubuntu 25.04 with the nightlies.
After a certain amount of time, the Device tab fails to load and displays a blank page when selected.
If I access it using a regular web browser (Chrome), the page loads reliably.

I'm able to restore the device UI by right-clicking the blank browser page and selecting 'Back'
Interestingly I've tried clicking 'Reload' as well but it has no effect.

NEOLabs-software, 2025-05-07:
Since I rethemed Orcaslicer, I was able to figure it out
Someone inserted a reload function when wxWidgets detects that you switch
out of it
I was able to fix it by simply removing that line of code, but for the
people who want to have a life, downgrade.
it would be great if softfever saw this issue and fixed it soon

norris108, 2025-05-07:
You could always create a pull request with your changes 😜

lodriguez, 2025-05-23:
all people running in this problem since April are using linux?

closed 2025-05-24 by Noisyfox

lodriguez, 2025-05-28:
> I'm on windows 11 using the V2.3.0_portable version and I'm now getting the white screen for the device tab. It was working fine last week.

Did you try the last [nightly](https://github.com/SoftFever/OrcaSlicer/releases/tag/nightly-builds)? This includes the patch.

brampat, 2026-09-15:
Having the same issue on Mac OSX 27.0
OrcaSlicer v. 2.4.2
Printer FlashForge Creator 5
On Prepare tab, I have my printer set up, If I click the WiFi icon and enter all details, then click Test, immediate successful connection (Connected to Flashforge local API successfully).
But Device tab stays blank. Right-click only ever shows Reload as an option there, but no Back or other options.
I do not know how Creator 5's can be reached through browser.
This issue does not appear on Flashforge's own software Flash Studio.

livibleu69, 2026-11-07:
Having same issue on win11 Orcaslicer 2.4.2 and 2.5RC (portable).
Orcaslicer connection test is successfull, even if when browsing no printer is discovered.
File print/upload is still working but the device sheet stays blank with "ERR_SOCKET_NOT_CONNECTED" or "ERR_CONNECTION_RESET" message.
However, Mainsail is working correctly from Edge or Chrome...