OrcaSlicer/OrcaSlicer #3835: Voluflow racetrack has weird changes in flow I didn't request

kozross, 2024-01-26:
### OrcaSlicer Version

1.9.0

### OS version

Arch Linux

### Additional system information

Nothing applicable

### Printer

Qidi XPlus-3 (custom)

### How to reproduce

1. Open the project
2. Slice the file
3. Observe weird flow inconsistencies

### Actual results

[image]

The picture speaks for itself. Notably, the first layer is done _ridiculously_ slowly, regardless of what flow I actually requested to start with (10), and the last bit is done at all kinds of weird flows (instead of the maximum of 40).

### Expected results

If I request a track from X to Y voluflow, I expect this to be honoured.

### Project file & Debug log uploads

[Max volumetric speed test.zip](https://github.com/SoftFever/OrcaSlicer/files/14060329/Max.volumetric.speed.test.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

TomPcz, 2024-01-26:
This is caused by "Extrusion rate smoothing". When disabled (set to 0), the first layer speed is as expected.

amdavidson, 2024-01-26:
> This is caused by "Extrusion rate smoothing". When disabled (set to 0), the first layer speed is as expected.

This is not the case on my install (1.9.0), you can see in this screenshot that both the weirdness on the top and bottom still exist and extrusion rate smoothing is set to 0.

[image]
[image]

TomPcz, 2024-01-26:
Oh, you are right. I encountered this (or quite similar) problem a few days ago. In my case, due to the very low speeds, the printer almost stopped for quite some time at the end of the first or beginning of the second layer. I was too busy to investigate it, but I managed to fix it (at least the first/second layer; not sure about the last one) by turning off the smoothing. However, I am unable to replicate it now. I do not think I have updated to a newer version since then.

kozross, 2024-01-26:
It seems that, if you have smoothing disabled _before_ you generate the model, this problem goes away. However, if you have smoothing enabled, generate, then disable it, it persists.

TomPcz, 2024-01-28:
I think I got it. It seems to be a combination of two things.

The flow changes (at constant speed) in the second and the last layer are caused by the spiral vase mode. In vase mode, the tool path is a helix, but the first and last layer are planes. To fill the gap between the helix and the planes correctly (without overextruding or leaving a gap), OrcaSlicer adjusts the flow linearly during the first and the last loop of the helix, while the speed is kept constant. Interestingly, SuperSlicer (2.5.59.6) or PrusaSlicer (2.7.0) do not do this. Instead, it seems that the first 3 layers are just printed planar, then it somehow transitions to the helix (which, I can imagine, causes overextrusion?) and the top seems to be kept helical. To me, what OrcaSlicer does makes sense but might be misleading during the flow test (on the last layer).

In addition, there seems to be a bug in the extrusion smoothing when spiral vase is enabled and bottom shell layers is 1 (I am trying to figure out if there is any other settings influencing this). In that case, the speed (and thus flow) are fluctuating during the print. This is, I think, what is causing also a very slow segment at the first layer of the flow test (as I said, my printer almost stopped for quite some time).

[image]

[image]

closed 2024-05-06 by github-actions[bot]