OrcaSlicer/OrcaSlicer #3838: Some filament options are not being saved

Primus007, 2024-01-26:
### OrcaSlicer Version

1.9.9

### OS version

Windows 11

### Additional system information

Intel(R) Xeon(R) CPU E3-1230 V2 @ 3.30GHz   3.30 GHz

### Printer

Elegoo Neptune 4

### How to reproduce

1. Open Filament Settings of an existing filament
2. Change nothing and save it under a new name
3. All options are the same as before
4. exit Orca Slicer
5. open Orca Slicer
6. Open Filament Settings and select the filament saved before

### Actual results

Now, many options have changed and no longer match the previously saved ones.
If I now change the print bed temperature for this filament, save it, and restart Orca Slicer, the print bed temperature is changed again. It retains some options, but others are reset after a restart.

### Expected results

The options set during saving should be retained after restarting Orca Slicer.

### Project file & Debug log uploads

[log.zip](https://github.com/SoftFever/OrcaSlicer/files/14065831/log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

seb449, 2024-01-26:
Hi, same pb, impossible to save profil !!!

mattnichols, 2024-01-27:
I'm seeing the same issue.

kylek29, 2024-01-27:
Curious, what is your Sync Profile in Preferences set to?

I used to have all sorts of problems like this until I disabled that setting. That was quite a few versions back though. IIRC, changing it changes where the configs are saved.

Primus007, 2024-01-27:
> Curious, what is your Sync Profile (there's a few settings in there that deal with syncing, iirc) in Preferences set to?
> 
> I used to have all sorts of problems like this until I disabled that setting. That was quite a few versions back though. IIRC, changing it changes where the configs are saved.

What do you mean with Sync Profile? And which setting do you have disabled?

Primus007, 2024-02-24:
any news about this? it is a bug, please fix it it is annoying.

Felix14-v2, 2024-05-25:
Go away

jbjhjm, 2024-08-15:
Confirmed from my side too. After I thought I had gone mad, this drives me mad.
Happens here for volumentric speed, nozzle temp and bed temp.
I set all temperatures to 0 to let my printer handle these.

I just checked the written json file. The temperature settings are simply not being persisted into the file at all. :(

jbjhjm, 2024-08-16:
Seems it has to do with inheritance of profiles (which I hate cause they are way too intransparent).
I edited the profile json files and emptied the "inherits" property.
Then, after restarting Orca, the profiles seem to be saved correctly.

chrisbdaemon, 2024-08-26:
Bump, having this problem too.

jbjhjm, 2024-11-25:
go away bot, still an issue

Felix14-v2, 2025-02-24:
There's a chance that the issue may be fixed in 2.3.0 with the filament database rework, can someone check it?

ohhai-kthxbai, 2025-05-20:
Still seeing this today (2.3.0)

BigFudgeDD, 2025-12-31:
Hi there,

i can confirm that the issue still persists. I use Orcer Slicer v2.3.1 atm, running on Linux Mint 22.2 (the AppImage version of Orca Slicer, if that is important).

It happens when I copy an existing filament: 
"Save current filament" --> give it a new name --> change some settings --> save it again under the new name --> close Orca Slicer --> restart Orca Slicer --> the new filament is completely f****d up. For example when I copy a PETG profile, the new filament is changed to PLA with wrong nozzle and bed temperature, fan speeds, max. volumetric flow and a bunch of other parameters. Only the name and the color is retained.

What I was able to troubleshoot so far is the following: When I go to the folder with my user data (in my case "/home/username/.config/OrcaSlicer/user/default/filament") and look at the files directly, there is a distinct difference between the working profiles and the problematic. Here are two examples:

problematic profile (PETG Jayo rot.json):

`
{
    "default_filament_colour": [
        "#ED333B"
    ],
    "filament_diameter": [
        "1.75"
    ],
    "filament_settings_id": [
        "PETG Jayo rot"
    ],
    "from": "User",
    "inherits": "PETG Jayo schwarz",
    "is_custom_defined": "1",
    "name": "PETG Jayo rot",
    "pellet_flow_coefficient": [
        "0.415752"
    ],
    "version": "1.6.0.0"
}
`

working profile (PETG Jayo schwarz.json):

`
{
    "activate_air_filtration": [
        "0"
    ],
    "activate_chamber_temp_control": [
        "0"
    ],
    "adaptive_pressure_advance": [
        "0"
    ],
    "adaptive_pressure_advance_bridges": [
        "0"
    ],
    "adaptive_pressure_advance_model": [
        "0,0,0\n0,0,0"
    ],
    "adaptive_pressure_advance_overhangs": [
        "0"
    ],
    "additional_cooling_fan_speed": [
        "0"
    ],
    "chamber_temperature": [
        "0"
    ],
    "close_fan_the_first_x_layers": [
        "2"
    ],
    "compatible_printers": [],
    "compatible_printers_condition": "",
    "compatible_prints": [],
    "compatible_prints_condition": "",
    "complete_print_exhaust_fan_speed": [
        "80"
[... 251 lines cut ...]
    "textured_cool_plate_temp_initial_layer": [
        "40"
    ],
    "textured_plate_temp": [
        "70"
    ],
    "textured_plate_temp_initial_layer": [
        "70"
    ],
    "version": "1.6.0.0"
}
`

So it seems there is some problem with this "inherits:" stuff. What is that even for? Just save the file with all parameters, a SSD in this day and age can handle a few kB per file.

For now, I found the following workaround:

Close Orca Slicer, go to the filaments folder in the user data, copy the *.json of a working profile (Ctrl+C; Ctrl+V) and change the name as you like. I copied the corresponding *.info as well, but I don't know if this is necessary. Afterwards, open Orca Slicer. The new filament shows up under the new name with correct settings (obviously a 1:1 copy of the working profile). Now you can change the settings within the new profile, save them under the new name and after a restart of Orca Slicer the changes are actually retained.

PLEASE fix this! It is extremely annoying and frustrating to have to copy and paste new filaments manually, just because you seem to be afraid of <6 kB file size in the year 2025/2026. Or at least tell us what we are doing wrong and how copying of an existing filament to a new one is supposed to work. But outright ignoring the issue for almost two years now isn't cool!

Thank you in advance

[PETG Jayo rot.json](https://github.com/user-attachments/files/24397469/PETG.Jayo.rot.json)
[PETG Jayo schwarz.json](https://github.com/user-attachments/files/24397468/PETG.Jayo.schwarz.json)

Felix14-v2, 2026-10-06:
Try this workaround: https://github.com/OrcaSlicer/OrcaSlicer/issues/3669#issuecomment-6019734534
Detaching should remove all links to the parent profile, so its removal shouldn't affect the new one in any way.