OrcaSlicer/OrcaSlicer #3841: FR: allow all speeds to be set as percentage of single centralized (base) value

jamincollins, 2024-01-26:
**Is your feature request related to a problem? Please describe.**
When trying to increase the speed a printer prints at, there are numerous settings that one must adjust within Orca Slicer.

**Which printers will be beneficial to this feature**
All

**Describe the solution you'd like**
It would be nice to be able to specify each speed value as a percentage of single centralized value, like say **base speed**.

**Describe alternatives you've considered**

**Additional context**

kylek29, 2024-01-27:
I'd like to add that it'd be nice to have the ability to override this base speed via Filament settings. Certain filaments require you slow down the printing speed, right now you have to manage different process settings for each.

With this change, you could set a PETG filament profile to override base_speed from 500 to 250, thus halving the speed.

An alternative approach would be to have a speed_modifier (%) variable that the user could set (again, in Filament and Process settings), and then be able to call that variable on each speed setting in process.

Speed_Modifier: 120% --> 1.2

------

Perimeters: 100 * speed_modifier 
Small Perimeters: 50

Results in:
- Perimeters: 120
- Small Perimeters: 50

m1shan, 2024-04-30:
This feature is a must!

closed 2024-05-18 by jamincollins