OrcaSlicer/OrcaSlicer #3847: Slow down 1st layer above bridge

AndreySemjonov, 2024-01-26:
Would like to have option to slow down 1st layer above bridge to make better contact between bridge and next layer.
If I have fast printing speed (200mm/s+) than I think this affect next layer quality and how bridge attaches to next layer.
Same function as Cura have - Bridge (Second/Third) skin speed (density, flow).

Tianziegler, 2024-02-24:
i would also be happy to have such a feature when printing a bridge it looks fine but as soon as the second layer gets printed the bridge starts to sag by a lot. Printing rPet at 260°C with a 0.5mm nozzle. Bridge speed 15mm/s and internal solid infill at 400mm/s. Would be good to be able to print two layer bridges for long distances or reduce the speed for the first layer ontop of the bridge and also be able to set a fan speed for that layer.

aranadam, 2024-05-19:
Please add this feature!!!

Ataraxia-Mechanica, 2024-05-26:
As an owner of a 500mm/s printer, please add this alongside slowing down for internal solid infill overhangs: #5003

Rodamyot, 2024-06-08:
I'd love to have this to. Sure I can do multiple layer seting and change speed but the goal in a slicer is to be able to identify trouble in printing parts and solve it. the surface over bridge is so so and speed printing or even normal speed printing need a different parameter to compensate for infill over bridge sirface, like more flow extrusion and slowing speed.

wbagnall, 2024-06-14:
I second this. An above-bridge speed setting is crucial for bridge quality. OrcaSlicer is unusable for bridging without this feature.

GabrieleCzz, 2024-06-16:
I also would love this feature!
I made some tests of bridging and one of them was comparing slow speed on top of the bridge w.r.t. normal speed (in both cases the bridge speed was slow). The difference was pretty noticeable!
So, to have good bridges without reducing overall speed (or make manual modifications) it's really necessary to have this feature at least on the first layer above the bridge (maybe also on the second one).

leokeba, 2024-07-12:
Yes please, this feature is very much needed, especially on fast printers when using low layer heights. My Bambu Lab P1S makes nice bridges but they get destroyed as soon as the next layer prints on top of them at full speed.

Bischi1997, 2024-07-17:
I would love to see this feature as well :)

stanpete78, 2024-08-17:
Please do this. On fast printers the bridges are so much worse thn the need to be.

cwiegert, 2024-09-23:
Yes Please, such a powerful feature for using today's high speed filaments

oleksii-suprun, 2024-09-24:
I have found a workaround and used the "Height Range Modifier" feature to slow down the infill over the bridge, which is inconvenient. Having a configuration option would be better.

jamincollins, 2024-10-01:
Just another user requesting this same feature.
:+1:

ph0tic, 2024-10-11:
yeah. me too pls.

ph0tic, 2024-10-11:
I do try to print as fast as possible, and i find when the external layer comes back across the bridge, it knocks off the edge on occasion. that really, really sucks.

igor-rubis, 2024-12-08:
+1 it would be great to have this feature

xateesix, 2024-12-10:
Not just great to have it, absolutely necessary.

ebrune81, 2025-01-01:
I've been struggling with this exact problem. Having to slow down the entire infill speed for just the layer above bridge layer seems crazy.

sutty5, 2025-01-05:
This feature is a must. Multi-layer bridge settings are needed. This is because, for those printing at high speeds (which is becoming more common), when a bridge layer is printed, the next layer on top of the bridge is shown as "internal solid infill". For which, I have mine set at high speed, which, is usually fine. However, for medium to long distance bridges, printing over the top of a bridge layer at 400mm/s and max flow rate completely screws the bridge layer. There needs to be a multi-layer bridge setting. Such that one can ensure the "solid infill" layer on top of a freshly laid bridge is printed slower, ensuring the bridge is properly set and reinforced before proceeded to speed over it with a high speed solid infill layer. I think it should go like so... Bridge Initial Layer, Bridge Final Layer, Internal Solid Infill.

Bromim, 2025-01-12:
I agree, adjusting the print speed of the layer on the bridge is necessary.

igiannakas, 2025-02-01:
Hi all

Some ground work has been done on this PR here: https://github.com/SoftFever/OrcaSlicer/pull/8263

There is of course plenty more to be done, but the "hard work" of identifying and segmenting the regions above bridges is now complete.

igiannakas, 2025-02-01:
**Test results from that PR:**
Left - Old / Right - Two bridge layers enabled
Bridge is significantly better
[image]

closed 2025-02-12 by SoftFever

stefansjs, 2025-02-26:
Can I request a change to the implementation? It seems like it's excluding long narrow bridges, where slowing down is more critical for me.

[image]
[image]

igiannakas, 2025-02-26:
@stefansjs You’re using classic and the bridge you’re seeing is not a bridge, but gap fill. Either make the extrusion width narrower to actually fit a bridge or reduce number of walls.

This bridge area is too thin hence gets filtered out by the algorithm too.

Don’t forget a bridge is coloured blue and it needs to be at least 1-2 lines wide for it to “count”.

igiannakas, 2025-02-26:
Can you post a screenshot of the layer below the second bridge layer? Ie the first bridge layer please?

stefansjs, 2025-02-26:
Here's the two adjacent layers. This happens for each bridging leg in the model.

[image]
[image]

My understanding from the tooltip explanation is that Orca Slicer refuses to generate an *extra* layer if the bridge is too narrow. "Please note that small bridges that are shorter or narrower than the set number of perimeters will be skipped as they would not benefit from a second bridge layer." It seems like this is intended functionality. I'm just hoping for a more permissive extra layer.

igiannakas, 2025-02-26:
Can you post the project file please?

stefansjs, 2025-02-26:
[bridging_test_50mm.3mf.gz](https://github.com/user-attachments/files/18982975/bridging_test_50mm.3mf.gz)

Should I open a new issue?

igiannakas, 2025-02-26:
Let me investigate first and will let you know

SilencedFrost, 2025-10-24:
same problem, my first bridge is looking really nice now, but on the second layer, the fast wall ruins it, having a slow down option for first layer above bridge would be wonderful

1st bridge layer

[image]

2nd bridge layer

[image]

igiannakas, 2025-10-25:
The surface area of the bridge does dictate whether you need a second bridge layer. This is used to filter out trivial bridges as they don’t tend to sag or break. 

I’ve tested this with Voron parts where the created bridges are small - for example the flat top of a screw hole. There adding a second bridge layer doesn’t do anything to improve quality as the bridge didn’t separate at all (as it’s too small.). If you have a specific part that is giving you trouble post a pic and I can post a PR that removes that constraint for you to test if it is beneficial to reduce the filtered area.