OrcaSlicer/OrcaSlicer #3852: Vertical shifted object positions when project uses multiple plates

phranck, 2024-01-27:
### OrcaSlicer Version

1.9.0

### OS version

macOS Sonoma 14.3

### Additional system information

Mac Studio M1 Max, 32GB RAM

### Printer

Bambu Lab X1C

### How to reproduce

1. Create a new project in Bambu Studio
2. Place objects on multiple plates
3. Save project, close Bambu Studio
4. Open the new saved project in Orca Slicer

### Actual results

All models beginning with the 5th plate (second row) are shifted vertically in Orca Slicer.

Project in Bambu Studio:
[image]

Same project in Orca Slicer:
[image]

### Expected results

All models should be at the same place in Orca Slicer as in Bambu Studio.

### Project file & Debug log uploads

[TestProject.3mf.zip](https://github.com/SoftFever/OrcaSlicer/files/14072402/TestProject.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

phranck, 2024-03-01:
No single feedback within over a month? 🤔

phranck, 2024-04-06:
Hello? Anybody out there...? 🤔

KevAquila, 2024-07-04:
Same issue here, really annoying.  Just adding to this that it still happens on latest version.

phranck, 2024-07-18:
Has anyone looked at this issue yet? I'm a bit confused that no-one is responding here. 🤔

andcolla80, 2024-08-02:
I'm having the very same issue.
Sometimes I need to reopen the file twice to get objects in the right position.  3MF files that I created and only use in Orca.
I realized the problem gets worse when change from a A1 Mini project to P1S project and the bed size changes. 
The shifted position is always on Y axis.

andcolla80, 2024-08-02:
First open:

[image]

Just re-open de same file:

[image]

I know when the position will be right if the "Unsaved changes" shows up. Otherwise it will be shifted.

an3k, 2024-09-27:
The space between the rows of plates is greater in OrcaSlicer than in Bambu Studio. Because the objects are saved in XY coordinates and are not rooted to an individual plate but the whole project, they stay at the same position while the position of the plates are shifted.

Easiest fix: make the space between the plates in OrcaSlicer identical to the space in Bambu Studio.
Ideal fix: root the coordinates for objects to their plate, not the whole project, that way the plate could be anywhere in space

sonjamichelle, 2024-11-20:
Since the devs decided to ignore and close my issue (#6434) I'll copy my last comment here:

NOW I am seeing that even though I created the 3MF WITH OrcaSlicer, when I load multiple plates 4 and beyond NO LONGER HAVE THE PARTS IN THE CORRECT LOCATION!

maddin8, 2025-01-02:
I'm experiencing the same and it seems to have gotten worse over the last few says. Been trying to open a 3mf for about the 10th time now and still seeing everything except the first plate shifted so that all objects are off those other plates. Bambu Studio opens and displays everything fine, as intended.

I'm no programmer but I'm happy to help if any diagnostic/log data is required, unfortunately I can't share the 3mf for copyright reasons and I don't have any other projects files that contain multiple plates :(

3dfanattic, 2025-02-09:
I also have this issue. Sometimes it shifts down, sometimes it shifts up.

The file was made in Orca Slicer and never used in Bambu Studio.
It's particularly annoying because it shifts my parts off the plate and I need to adjust to compensate every time I print these plates.

Every time I save the plate with the proper placement, and then open it again later, everything is shifted down.

[image]
[image]

parrothd, 2025-02-16:
I have the same issue, the objects on 2nd/3rd row plates always move up. Move them to the correct spot save and reopen and the objects have moved.

closed 2025-05-25 by github-actions[bot]