OrcaSlicer/OrcaSlicer #3856: "Access violation" after mirroring rotated and cloned object

mikesmitty, 2024-01-27:
### OrcaSlicer Version

1.9.0

### OS version

Windows 10

### Printer

Voron V2.4 0.4 nozzle

### How to reproduce

1. Insert STL (tried with two pretty simple models, one just a rectangular solid block)
2. Use Lay on Face, select either left or right sides of the box (front/back/top don't cause it to happen)
3. Mirror along X or Z axis (Y doesn't cause it)
4. Slice plate

### Actual results

A fatal error occurred: "Access violation"
Please save your project and restart the application.

[image]

### Expected results

Model is sliced

### Project file & Debug log uploads

[box - access violation.zip](https://github.com/SoftFever/OrcaSlicer/files/14073213/box.-.access.violation.zip)
[log.zip](https://github.com/SoftFever/OrcaSlicer/files/14073217/log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

larsirion, 2024-02-10:
Same error here with a Creality K1 but along Y axis and no error along X axis (thanks for the hint by the way it helped me today as it appears the bug is only along one of the plane axis).

exzhawk, 2024-03-25:
Same erro with version 2.0.0-rc, Windows. Creality K1C.
reproduce step
1. new project
2. import model
3. lay on face (select the front face)
4. right click - clone 1
5. right click the cloned one - mirror along y axis
6. shift+a to arrange
7. slice plate
[debug_Mon_Mar_25_22_35_04_25604.log.zip](https://github.com/SoftFever/OrcaSlicer/files/14744994/debug_Mon_Mar_25_22_35_04_25604.log.zip)

Roman-Golod, 2024-04-15:
Have the same. Orca 2.0.0

jeffhengesbach, 2024-05-08:
> Have the same. Orca 2.0.0

Ditto

Bastlwastl84, 2024-06-02:
same Error after mirroring  in Y in 2.1.0 beta
[Leiste seitlich.zip](https://github.com/user-attachments/files/15526144/Leiste.seitlich.zip)

lennoxdk, 2024-06-13:
same error mirror and rotate

BETLOG, 2024-06-16:
+1
lay on face (affects a 270 degree Y axis rotation)
right click, clone, +1
move clone a few mm away from original in ~X axis
mirror (clone), X axis
dolly/pan/zoom view, deselect clone, or leave it selected... seems to make no diff
slice (tab - which is extremely unreliable in linux BTW, pressing tab to slice may or may not happen ~50% of the time)
app exits silently

[2024-06-17--09-10-40_konsole+log.zip](https://github.com/user-attachments/files/15860112/2024-06-17--09-10-40_konsole%2Blog.zip)
Kubuntu 24.04
OrcaSlicer_Linux_V2.0.0-beta.AppImage
OrcaSlicer_Linux_V2.0.0.AppImage
OrcaSlicer_Linux_V2.1.0-alpha.AppImage

closed 2024-09-22 by github-actions[bot]