OrcaSlicer/OrcaSlicer #3857: Orca Slicer Hangs when opening 3mf project

FedorSosnin, 2024-01-27:
### OrcaSlicer Version

2.0

### OS version

macOS Sonoma 14.2.1

### Additional system information

2021 MacBook Pro M1 Max + 64gb memory

### Printer

Creality K1 Max

### How to reproduce

1. Open 3MF file

### Actual results

Orca Slicer Hangs at about 3/4 of the way and stops responding 

### Expected results

To fix the issue, I tried re-installing, clearing out the configuration folder and copying over to a new install, rebooting the machine, and Installing the nightly build with no luck. 

I was able to get the 3mf project open by **logging out** of the logged-in user.  That solved the issue. 

Hope this helps. 

### Project file & Debug log uploads

[debug_Fri_Jan_26_21_22_07_65037.log.0.zip](https://github.com/SoftFever/OrcaSlicer/files/14073349/debug_Fri_Jan_26_21_22_07_65037.log.0.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

FedorSosnin, 2024-01-29:
Just encountered the problem again, this time I was signed out. Ran through the same troubleshooting with no luck. Only fix was to sign back in. So something related to sign in/out seems to solve the issue.

klasnet21, 2024-05-01:
merhaba neden orca slıcer ile 3mf dosyalarını açamıyorum

closed 2024-08-09 by github-actions[bot]