OrcaSlicer/OrcaSlicer #3860: Calibration-Advanced Volumetric Flow

grandoth, 2024-01-28:
### OrcaSlicer Version

1.9.0

### OS version

Windows 10

### Printer

Creality K1 Max

### How to reproduce

1. Create a New Project.
2. menu Calibration -> Max Flowrate -> Accept Defaults (5, 20, 0.5) (OK)
3. View the Layer height settings under the Quality tab.

### Actual results

The "Layer height" and "First layer height" settings are changed from my default for the system presets to 0.32 mm. Following the Calibration guide, it suggests using the expression which implies a 0.2 mm step height:
start + (height-measured * step)

with example 5 + (19 * 0.5) , or 13mm³/s

When I print this test model, the height is in fact 32 mm.

[image]

### Expected results

The "Layer height" and "First layer height" settings should both be set to 0.2 mm step height OR the wiki Calibration page described above should be updated to account for the measurement of 32 mm.

### Project file & Debug log uploads

[Max volumetric speed test.zip](https://github.com/SoftFever/OrcaSlicer/files/14074902/Max.volumetric.speed.test.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

SoftFever, 2024-01-30:
The model is printed in vase mode, hence it's not an issue. The layer height is purposely set to 0.32 to avoid increasing the printing speed too quickly.

grandoth, 2024-01-30:
OK. I see now that the model automatically calculates its height depending on the step size and end max volumetric speed. I think what might have been confusing me is the following from the wiki: "5 + (19 * 0.5) , or 13mm³/s" when the result of that expression evaluates to 14.5. Or am I misunderstanding the calculation?

closed 2024-05-07 by github-actions[bot]