OrcaSlicer/OrcaSlicer #3871: Slicing hangs on tree slim slice

Alattice, 2024-01-28:
### OrcaSlicer Version

1.9.0

### OS version

Windows 10 22H2 build 19045.3930

### Additional system information

The slicer will hang when using tree slim supports. Although the program will still respond, attempting to stop the slicing will freeze orcaslicer. And a spike in cpu and ram usage occurs and stays high.

CPU: Ryzen 5 5600
RAM: Corsair Vengnece 32GB
[image]

### Printer

QiDi X-Plus 3

### How to reproduce

1. paint on supports to a model
2. in supports (global or object), enable support and select Tree (manual)
3. set style to Tree Slim
4. slice plate
5. slicing percent will get stuck at some point (65% for me)

### Actual results

slicing gets stuck at a percentage and both ram and cpu spike up

### Expected results

models get sliced and completed with tree support

### Project file & Debug log uploads

[project-file.zip](https://github.com/SoftFever/OrcaSlicer/files/14076747/project-file.zip)
[debug_Sun_Jan_28_11_18_33_19968.log.0.zip](https://github.com/SoftFever/OrcaSlicer/files/14076791/debug_Sun_Jan_28_11_18_33_19968.log.0.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

closed 2024-05-06 by github-actions[bot]

JK0JK, 2024-06-19:
This is still a problem and also happens with Tree Slim on Tree(Auto). This is also a bug in Bambulabs.

Sidiusz, 2024-09-12:
It's not STALE at all! It happens with Strong Trees too.