OrcaSlicer/OrcaSlicer #3872: Filamenttype for all Printers

Slevin7, 2024-01-28:
Orca Slicer has an extremely large number of filaments as presets in the slicer itself.
Unfortunately, only if you have selected all printers.
Since I only have one printer, and I don't think I'm alone in this, only the Creality Filmante are displayed.
It would be great if I could see them all. 
[image]
[image]

atl285, 2024-01-30:
If you don't have a Bambu Lab printer configured, it makes no sense to show you the other filament profiles, because they are calibrated especially for the Bambu Lab printers.

So for all printers, which has no pre-configured (calibrated) filament profiles, the Generic profiles are created.

Slevin7, 2024-01-30:
I see, I thought these were the starting points for each filament, like preferred temperature.
I can then use them to adjust my stuff, like pressure and temp and so on.

closed 2024-05-07 by github-actions[bot]