OrcaSlicer/OrcaSlicer #3874: Custom Filament - Choosing Custom Filament in AMS slot (Device page) won't save/persist after clicking 'confirm'

Vantage-DS, 2024-01-28:
### OrcaSlicer Version

1.9.0

### OS version

Mac Sonoma 14.2.1 

### Additional system information

Mac Mini M1 16GB

### Printer

Bambu Lab P1S (LAN Mode)

### How to reproduce

LAN mode - Haven't tried cloud/Don't want to use cloud.

1. Create custom filament
[image]

2. Map custom filament to AMS slot and click 'confirm'
[image]

3. Continue to map to other filament slots 
[image]

4. Check back on filament slot - Filament selection is blank, although 1 of the AMS slots (seems random) may actually save the info.
[image]

### Actual results

After mapping the filament to the AMS slots, going to the Prepare page syncing the filament, the below happens -
[image]

Filaments are shown as generic because the custom filament wasn't saved in the AMS filament mapping.

### Expected results

Create custom filament.
Map custom filament to AMS.
Sync filament.
Correct filament is shown on the Prepare page.

### Project file & Debug log uploads

[log.zip](https://github.com/SoftFever/OrcaSlicer/files/14077356/log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

CryPt00n, 2024-02-05:
Same for spool slot.

CryPt00n, 2024-03-19:
@Vantage-DS I found a solution for this. You have to enable preset sync with your Bambu Account. Then, custom filaments will be synced to the printer & correctly shown on the device page.

[image]