OrcaSlicer/OrcaSlicer #3875: Add more nozzle profiles for Prusa MK3S Profile.

BladerunnerxRC, 2024-01-28:
**Is your feature request related to a problem? Please describe.**
Only one nozzle size. A 0.4 mm profile is available for Prusa MK3S.

**Which printers will be beneficial to this feature**
Prusa MK3S

**Describe the solution you'd like**
Please add the following nozzle sizes: 0.2, 0.6, 0.8

**Describe alternatives you've considered**
The alternative is to use Prusa Slicer

**Additional context**
[image]

lovelace, 2024-02-12:
I'll add a vote for this one. My MK3S+ has the E3D Revo hot end installed and I regularly use all 4 nozzle sizes. Someone suggested I look into switching from PrusaSlicer but without the ability to use other nozzle sizes this is a complete non-starter for me. So, please add this feature. Thanks.

BladerunnerxRC, 2024-02-12:
Figured I was not the only one wanting this. It would be nice to stay in
the same slicer since I have an X1Carbon as well.

PR #4775 (2024-03-30): Various profile improvments 
Close #2286, close #3875, and partially #4761

Add Prusa MINI profiles (0.25,0.6,0.8), add MK3S+ nozzle types to generic filament types, & fix missing covers for Vzbot
closed 2024-03-30 by SoftFever