OrcaSlicer/OrcaSlicer #3876: Printbed display wrong for delta printers

bbinnard, 2024-01-28:
### OrcaSlicer Version

2.0.0-Dev

### OS version

Windows 11/64

### Printer

FLSun SuperRacer

### How to reproduce

This screenshot shows what the program shows with a simple flat disk loaded:

[image]

NOthing bad happens when slicing or uploading GCode to printer. It just looks bad on the monitor screen.  This problem cropped up with (I think) Ver. 1.9. 

### Actual results

Everything works OK

### Expected results

The bed display should look like it used to

### Project file & Debug log uploads

[projectbb.zip](https://github.com/SoftFever/OrcaSlicer/files/14077413/projectbb.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

PeterBeklemishev, 2024-02-02:
Duplicate of https://github.com/SoftFever/OrcaSlicer/issues/3581

bbinnard, 2024-05-03:
I got around the problem by making an STL file the same size as my print bed But as far as I can tell the basic problem still exists.

closed 2024-08-09 by github-actions[bot]