OrcaSlicer/OrcaSlicer #3880: Prime tower issues

KA009onPC, 2024-01-28:
### OrcaSlicer Version

1.9

### OS version

Windows 10

### Printer

Neptune 4 PLUS

### How to reproduce

i am trying to do a multi color print on my Neptune 4 plus. i have my M600 set up.

When i use prime tower and switch filaments and it goes to start printing and goes over to the prime tower. it just sits there and extrudes then it will start to move. i have the setting purge in prime tower turned off and I have purge in supports and infill turned off as well.

### Actual results

the machine just sits in the prime tower and extrudes and then finally moves.

### Expected results

this should let it use the prime tower properly 

### Project file & Debug log uploads

there are not debugging logs

### Checklist of files to include

- [ ] Log file
- [ ] Project file

KA009onPC, 2024-02-02:
> I have the same problem. is it already being looked at?

Hello, I havnt heard from anyone about it yet, you're the first

Tollie46, 2024-02-02:
@KA009onPC I have disabled it everywhere I can find it, maybe there is still a setting somewhere. where can i find this anywhere

KA009onPC, 2024-02-03:
Sadly, that's what I'm trying to figure out as well.. I've disabled everything.

PhilBaz, 2024-02-03:
I had this happen too...Single extruder multi material with manual color swaps.

It was a setting issue. It would go to the tower and make a big blob..

Find all the multi material settings and turn them to zero. I forget what setting it was. but yeah, turn everything to zero if you aren't using it or don't know what it does. The system is designed for bambu with purging so have to turn all that off.

KA009onPC, 2024-02-05:
I will give this a try thank you!

Tollie46, 2024-02-06:
@KA009onPC  I think I have found the problem, if you select multiple filaments the filament flushing volumes tab will appear above, I think if you set everything to 0 the purge is gone. I haven't been able to test this myself yet, but this will probably solve the problem
[image]

PhilBaz, 2024-02-06:
@Tollie46  Yes! that's what it was...set those to zero. that jogged my memory. Exactly.

Tollie46, 2024-02-21:
I still have the same problem but can't find it @SoftFever

Bad-Hackerman, 2024-03-31:
I am currently facing the same issue, found this in the printer settings:
[image]
I'm gonna play around with the settings to find out what works best.

PhilBaz, 2024-03-31:
@Bad-Hackerman Tolie46 posted the solution above... need to zero the flushing amounts.

Bad-Hackerman, 2024-03-31:
No, it's part of the solution, all the weird behavior comes from those settings above that are made for Bambu printers.

dohlin, 2024-04-18:
I too am seeing this issue...literally causing my small prints to fail.  Big blob is being created at the corner of the purge tower after moving to it and while switching filaments.  Why doesn't Orca do something similar to PrusaSlicer where the toolhead moves back and forth during the filament swap?  That solves the problem of releasing any pressure so seemlessly.

closed 2024-07-27 by github-actions[bot]

K-pah, 2025-01-23:
I stumbled upon this problem today, I know it's a little late, but I believe I figured it out.
I reckon that just like me, you were loading the filament manually before resuming the print.
Upon reading the g code that is generated by orca, it is commented that it also loads the filament (when it's standing still and pooping in place a 3 mm blob).
I believe it is not meant to load the filament manually before resuming and letting the gcode take care of that step, essentially only doing it once instead of twice.