OrcaSlicer/OrcaSlicer #3882: Can't switch off of dark mode on linux

sam0622, 2024-01-28:
### OrcaSlicer Version

1.9.0

### OS version

Linux Mint 21.2 Cinnamon

### Printer

Custom Klipper profile (for Ender 3 V2)

### How to reproduce

Go to settings, look for theme button.

### Actual results

The dark mode/ theme button is missing which is really annoying because all the grey makes some of the icons near invisible
I also went in the config file and changed it but it just set itself back to dark mode..
[image]

### Expected results

There should be a checkbox/ selector to change the theme.

### Project file & Debug log uploads

[orcaproject.zip](https://github.com/SoftFever/OrcaSlicer/files/14077937/orcaproject.zip)
[log.zip](https://github.com/SoftFever/OrcaSlicer/files/14077949/log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

closed 2024-05-06 by github-actions[bot]

vontrapp, 2024-05-28:
still a problem in 2.0.0 appimage

whytf, 2024-10-28:
Still a problem with 2.2.0

sperdev54, 2025-07-01:
> Still a problem with 2.2.0

still a problem with 2.2.0

barbaracleaver1915-star, 2026-05-16:
Still persists on Linux Mint Cinnamon 22.3 with OrcaSlicer 2.3.2.