OrcaSlicer/OrcaSlicer #3891: Slow Down For Overhangs - Apply to more than just the outer perimeter.

MandicReally, 2024-01-29:
Current Feature Behavior:
Currently the "Slow Down for Overhangs" feature works but it only applies the speed adjustment to the outer wall of the overhang. This leaves the inner geometry at the higher speed of the rest of the print and results in poor quality results. 

Feature Request:
It would be nice if there was some way to apply the Slow Down speed to more walls of the overhang. Maybe with a depth setting similar to "Shell Thickness". So maybe you could say that 1.3mm of wall thickness should be slowed down.

Current Workaround:
Right now I have to reduce inner wall & solid infill speeds to compensate for the areas that require the slow down. Usually this is achieved with a Height Range Modifier as only certain sections need to be slowed down, which is where the feature request comes in. The majority of a print doesn't need the inner wall slowed down, but overhangs do.

Here is a current Slicer Preview of the Slow Down for Overhangs doing its job, but the inner perimeters are still at high speeds:
[image]

And the Print Results of that fact:
[image]

igiannakas, 2024-01-30:
Enable slow down for curled overhangs. It evaluates the probability of an edge curling up internally as well and adjusts inner line speeds too.

MandicReally, 2024-01-30:
It does more than just the "Slow Down For Overhangs" but still not what I'd consider enough. Maybe that needs some controllable parameters? 

Same part with "Slow Down for Curled Perimeters" turned on:
[image]
At 2.6mm on this part with the exterior perimeter slowing down to the 75%-100% value of "Overhang Speed", the interior still isn't affected.

[image]
The "Slow Down for Curled Perimeters" starts at the 2.8mm layer here, but doesn't REALLY slow down until 3.2-3.4mm (screen grab is 3.4mm). By that point on the physical part the curling as well started already.

This is something I wasn't paying enough attention to, so the "Slow Down for Curled Perimeters" will help in this particular instance. I think I tried it and saw little real results (on previous parts) so I stopped paying attention to it. Still feels like some things would benefit from some more control of how it is implemented. None of it is a replacement for better part design, but I can't always control the designs of others.

igiannakas, 2024-01-30:
The idea here is that curling builds up and the model is trying to reduce it as it accumulates. Early layers shouldn’t have any or have very little internal speeds are not affected as much. As the curling up accumulates or gets more probable it slows down more and more.

the overhang speeds directly control the amount of slowdown for curling perimeters. If you want to make it more sensitive you can reduce the speed for the 75% overhang, which should make this more active. Curling is applying an artificial “overhang” distance to the existing perimeters internally.

MandicReally, 2024-01-30:
I will play with that feature more and see if I can generate the results I'm looking for. It would be great to not have to manually adjust (and slow down an entire area of a print just for one curled section), so I'll dive in a bit more as I get time.

faith-healer, 2024-05-01:
up for not to be stalled.

closed 2024-08-09 by github-actions[bot]

ShaneDelmore, 2024-12-01:
I ran into this myself today.  Curled perimeters option didn't help either.  So just another vote that this is a useful feature request.

igiannakas, 2024-12-01:
The issue here is this:
1. The current algorithm takes each perimeter at a time and identifies whether it’s overhanging. It calculates the overhang % and applies the speed modifier as per the overhang speed adjustment to the overhanging segment. As the internal perimeter is not overhanging (it’s fully supported by the ones underneath it) it’s not slowed down
2. It then looks at the layers below to check whether a sharp edge exists and to calculate the overhang % of the layers underneath. If both are exceeding a threshold it slows them down further (curled overhang detection).
3. It then looks at the proximity of an internal perimeter to an external perimeter that is identified as curled (point 2 above) and applies a % speed reduction based on how severely “curled” the external perimeter is. 

We could look at making point 3 a bit more sensitive or point 2 a bit more sensitive to trigger internal perimeter slowdowns a bit more, as in identify more perimeters as potentially curled.

 Currently it runs some pretty complex math in identifying the fluid dynamics behaviour of a filament when extruded to classify what may curl and how much. For example a sharp corner will curl up as the filament is elastic and tends to shrink when laid down over a sharp edge that is also partly overhanging. We could look at tweaking those thresholds to make them a bit more conservative.

for most cases the current approach should work well enough - however some more elastic material (like silk PLA) or when cooling is not as strong to freeze the material in place can result in more curling than anticipated by the mathematical model.

mattk926, 2025-05-08:
Just recently switched to orcaslicer and am running into this issue on latest 2.3.0 release this still appears to be a problem. This has been driving me mad trying to figure out why these 45degree overhangs on gridfinity bins weren't triggering my overhang fan setting. Any updates from the team on when this will be fixed or if this has worked in a previous version which version that is?

sebsucmor-alt, 2026-08-29:
[image] I added this on my fork, will be up on the 2_45 version in a week 
https://github.com/sebsucmor-alt/OrcaFS-NeotkoCM