OrcaSlicer/OrcaSlicer #3893: Print/Send dialog hangs and thread(s) goto 100% on  larger models.

nick-laird, 2024-01-29:
### OrcaSlicer Version

1.9.0 and greater (also happens in nightly)

### OS version

Linux

### Additional system information

Two systems both running Linux, one Arch other Ubuntu.
One is a Celeron 4x core. Other is a Xeon 8x core.
Celeron has 8GB RAM, Xeon has 64GB.

### Printer

Bambu X1 Carbon

### How to reproduce

1. Get a fairly large model or models that will take a bit of time to slice.
2. Slice them so the resulting file is over 10MB
3. Try to direct send or print to printer using the network plugin.
4. It works in 1.8.1 and freezes in later versions as the threads get stuck @ around 100%.
5. sometimes I can click cancel and wait and eventually the dialog closes. other times it doesn't and I have to force terminate the app.

### Actual results

[image]

### Expected results

The dialog should stop saying "Click Here if You Can't Connect" and should offer AMS, Bed leveling, etc checkboxes. Clicking Print should then start the transfer to the printer.

### Project file & Debug log uploads

This appears to be one of hte possible issues. After it does a PrintStatusInit() after I clicked "Print" it starts dumping:
2024-01-29 11:19:27.059810[Thread 0x00007003c3945480]:The values of tray_info_idx and tray_type do not match tray_info_idx  tray_type  system_type
Over and over again. Log attached.
[log.txt](https://github.com/SoftFever/OrcaSlicer/files/14086940/log.txt)

### Checklist of files to include

- [X] Log file
- [ ] Project file

erpalma, 2024-01-31:
I can confirm this issue. Ubuntu 22.04, tested 1.9.0 and nightly.

CCS86, 2024-03-10:
Same on Windows 10 x64

CCS86, 2024-03-10:
I reported here: https://github.com/SoftFever/OrcaSlicer/issues/3468

nick-laird, 2024-06-09:
Still exists in the 2.1.0-dev nightlies.

mjrider, 2024-08-04:
also present in linux mint wilma
trying to print a simple cube works fine, but a larger/complexer print hangs
highlighted the fields which are strange
[image]

model that triggered it https://www.printables.com/model/936619-dice-tower

real-artswan, 2024-09-11:
confirm with 2.1.1 Ubuntu 22.04.4 LTS, somehow larger models makes the dialog hanging

tak-21, 2024-10-20:
Same problem here.

- Linux mint v22
- OrcaSlicer v2.1.1
- OrcaSlicer Ubuntu2404_V2.2.0-rc

When I slice one "ghost" (see attached stl) the "Send print job to" dialogs just works.
If I clone the object, hence add a second ghost the dialog "Send print job to" hangs/is not usable.

[Ghost_for_LED_light.stl.zip](https://github.com/user-attachments/files/17449703/Ghost_for_LED_light.stl.zip)

infinitewarp, 2025-01-20:
Issue is not stale. This is still happening on the latest release (Linux V2.2.0 AppImage) on latest Pop!_OS 22.04 LTS. However, I also think this issue is a duplicate of #5531 .

real-artswan, 2025-01-24:
btw, I found that enabling Multi-device tab and sending prints as jobs might be a workaround

Beep6581, 2025-02-23:
Still reproducible in v2.3.0-beta.

I extracted the AppImage, ran the executable through GDB and broke when it hung:
```
^C
(gdb) bt
#0  std::basic_streambuf<char, std::char_traits<char> >::sgetc (this=0x7fffffff51e0) at /usr/src/debug/gcc/gcc-build/x86_64-pc-linux-gnu/libstdc++-v3/include/streambuf:349
#1  std::operator>><char, std::char_traits<char>, std::allocator<char> > (__in=..., __str="") at /usr/src/debug/gcc/gcc/libstdc++-v3/src/c++98/istream-string.cc:67
#2  0x0000555557183d25 in Slic3r::GUI::SelectMachineDialog::is_show_timelapse() ()
#3  0x0000555557192e55 in Slic3r::GUI::SelectMachineDialog::update_select_layout(Slic3r::MachineObject*) ()
#4  0x000055555719f400 in Slic3r::GUI::SelectMachineDialog::on_timer(wxTimerEvent&) ()
#5  0x0000555559b40591 in wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&) ()
#6  0x0000555559b4090e in wxEvtHandler::SearchDynamicEventTable(wxEvent&) ()
#7  0x0000555559b40a74 in wxEvtHandler::TryHereOnly(wxEvent&) ()
#8  0x0000555559b40b1f in wxEvtHandler::ProcessEventLocally(wxEvent&) ()
#9  0x0000555559b40bc9 in wxEvtHandler::ProcessEvent(wxEvent&) ()
#10 0x0000555559b4154b in wxEvtHandler::SafelyProcessEvent(wxEvent&) ()
#11 0x0000555559b1c9e4 in wxTimerImpl::SendEvent() ()
#12 0x0000555559a5a2ef in timeout_callback ()
(...)
(gdb) c
Continuing.
(...)
^C
(gdb) bt
#0  0x00007ffff7cc9acd in std::ctype<char>::scan_is (this=<optimized out>, __m=8192, 
    __low=0x5555e3185d16 "532 E.0074\nG2 X142.19 Y184.335 I9.367 J-5.819 E.0269\nG1 X142.189 Y184.603 E.0074\nG3 X141.64 Y185.406 I-9.367 J-5.819 E.0269\nG1 X141.641 Y185.673 E.0074\nG2 X142.19 Y186.476 I9.367 J-5.819 E.0269\nG1 X14"..., __high=0x5555e31877bf "@ ") at /usr/src/debug/gcc/gcc-build/x86_64-pc-linux-gnu/libstdc++-v3/include/x86_64-pc-linux-gnu/bits/ctype_inline.h:60
#1  std::operator>><char, std::char_traits<char>, std::allocator<char> > (__in=..., __str="") at /usr/src/debug/gcc/gcc/libstdc++-v3/src/c++98/istream-string.cc:79
#2  0x0000555557183d25 in Slic3r::GUI::SelectMachineDialog::is_show_timelapse() ()
#3  0x0000555557192e55 in Slic3r::GUI::SelectMachineDialog::update_select_layout(Slic3r::MachineObject*) ()
#4  0x000055555719f400 in Slic3r::GUI::SelectMachineDialog::on_timer(wxTimerEvent&) ()
#5  0x0000555559b40591 in wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&) ()
#6  0x0000555559b4090e in wxEvtHandler::SearchDynamicEventTable(wxEvent&) ()
#7  0x0000555559b40a74 in wxEvtHandler::TryHereOnly(wxEvent&) ()
#8  0x0000555559b40b1f in wxEvtHandler::ProcessEventLocally(wxEvent&) ()
#9  0x0000555559b40bc9 in wxEvtHandler::ProcessEvent(wxEvent&) ()
#10 0x0000555559b4154b in wxEvtHandler::SafelyProcessEvent(wxEvent&) ()
#11 0x0000555559b1c9e4 in wxTimerImpl::SendEvent() ()
#12 0x0000555559a5a2ef in timeout_callback ()
```

3vi1, 2025-03-15:
@SoftFever :  I see the issue:

is_show_timelapse is scanning the gcode for "BambuStudio 01.04.##...." or higher (to make sure it supports the feature?).  Normally this would be on line 2 of the gcode ("generated by OrcaSlicer 2.3.0-rc ...") so the function doesn't break out of the read loop.  That's what makes it evident only with very large models.

I changed standard_version on line ~3666 to be "2.2.0" and line 3684 to "OrcaSlicer" from BambuStudio, so that it will pass the check if the gcode was generated with OrcaSlicer 2.2 or higher.  After that, the dialog is fully responsive again.

SoftFever, 2025-03-16:
> [@SoftFever](https://github.com/SoftFever) : I see the issue/fix:
> 
> is_show_timelapse is scanning the gcode for "BambuStudio 01.04.##...." or higher (to make sure it supports the feature?). Normally this would be on line 2 of the gcode but OrcaSlicer puts its own info there ("generated by OrcaSlicer 2.3.0-rc ...") so the function isn't breaking out of the read loop there as it should. That's what makes it evident only with very large models.
> 
> I changed standard_version on line ~3666 to be "2.2.0" and line 3684 to "OrcaSlicer" from "BambuStudio", so that it will pass the check if the gcode was generated with OrcaSlicer 2.2 or higher. After that, the dialog is fully responsive again.

Wow, thank you so much for the troubleshooting!
Will you submit your fix?

3vi1, 2025-03-16:
> Wow, thank you so much for the troubleshooting! Will you submit your fix?

@SoftFever :  Sure thing.  I'll get it submitted tomorrow!

SoftFever, 2025-03-16:
> [@SoftFever](https://github.com/SoftFever) : Sure thing. I'll get it submitted tomorrow!

Thank you.
Merged 👍

closed 2025-06-23 by github-actions[bot]