OrcaSlicer/OrcaSlicer #3896: Crash when attempting to edit filament under Device

pgifford, 2024-01-29:
### OrcaSlicer Version

2.0.0-dev

### OS version

Windows 10

### Printer

Bambu X1C

### How to reproduce

1. Click on "Device"
2. Click on pencil icon over filament

[image]

### Actual results

Error dialog, program closes

[image]

### Expected results

I can edit filament setting on device tab.

### Project file & Debug log uploads

[project_and_log.zip](https://github.com/SoftFever/OrcaSlicer/files/14090676/project_and_log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

KrisMorr, 2024-01-30:
I confirm, I also have the same error
[image]

pennystockss, 2024-02-03:
Same on Mac, tried various builds. Orca just closes no errors.

pgifford, 2024-02-11:
Issue no longer occurs with latest dev, fixed by https://github.com/SoftFever/OrcaSlicer/pull/3981 (probably)

closed 2024-02-11 by pgifford