OrcaSlicer/OrcaSlicer #3899: Orca Slicer slows down computer

MYexeto, 2024-01-30:
Hi there, 
When the slicer is open, my computer starts lagging like a hell, specially when CAD software + Orca Slicer are running together.
But in the task manager there is no load on computer at all...
Once close Orca Slicer, everything comes back to normal.

Orca Slicer V1.9.0. 

Same thing with Bambu studio, but not with prusa slicer...

CPU intel i5 13600k
RAM DDR5 32gb 5600
GPU RTX 4070ti

AndreySemjonov, 2024-01-30:
I confirm. I found same yesterday.
I couldn't understand why everything so slow. Lagging etc.
In task manager no such load.
After I close Orca Slicer all went good. But it was open for few days in background.

CPU i7 12700k
RAM DDR4 32gb
GPU RTX 3060ti

Adanufgail, 2024-01-31:
Can confirm. My issues were primarily in Outlook/Explorer. I had performance monitor running and saw no uptick in IO, CPU, RAM, or SWAP usage (RAM was comfortably at below 50% utilization so SWAP shouldn't even be a factor). Restarted multiple times, which seemed to help temporarily (obviously until I opened Orca back up to continue working on projects and monitoring the printer). Ran SFC/DISM multiple times, installed Windows updates, installed Office updates.

CPU AMD Ryzen 5 5600X
RAM DDR4 32 GB (4x8GB)
GPU NVIDIA GeForce RTX 3060 TI Eagle OC D6X 8G

MYexeto, 2024-02-01:
> What OS versions are you guys running?
> I haven't noticed this on my Windows 10 machine with an Nvidia GFX card.
> Can confirm. My issues were primarily in Outlook/Explorer. Changing
> directories in Explorer or any file operation would lag for 5-10 seconds.
> In Outlook, clicking an email, hitting Reply/Reply All, etc. would all lag
> 10-30 seconds per item.
> Other troubleshooting: I had performance monitor running and saw no uptick
> With Orca Slicer closed, no issues whatsovever going on 5 hours now.
> Just as a side note .. although you may have an idea of the culprit in
> this case, in the future one thing you might try running is LatencyMon
> (free 3rd party tool), it's useful in helping identify driver/software
> issues that don't show up in performance monitor (e.g. no high RAM/CPU
> usage). You'd look for very high ISR/DPC timings and the associated
> processes.

Win 11

asiersan, 2024-02-24:
same here with Windows 11 Pro
 i7-6700K 
16GB Ram

fen4ko, 2024-03-01:
Ideamaker for conveyor belt printer and the orca for everything else, I started noticing slowness to the point of checking for viruses, since ram were not the problem I finally closed the orca and the lag is gone (The lag is very noticeable from the moment you click on the submenus)

thinking that the problem was the included web viewer and I have closed the process in task manager without any improvement

Orca Slicer 1.9.1
w11pro 
i7 6700HQ 
32gb ram
Nvme samsung 970 
gf Gtx 960m

[image]

dluzh, 2024-03-02:
Exactly same problem

Orca Slicer v1.9.0

Windows 11
Intel COre i7-7500
16GB RAM
WD Red 500Gb SSD

MrDavidFantasy, 2024-03-02:
I confirm that the computer is very slow. After closing Orca Slicer everything returns to normal operation. Especially noticeable when the slicer is open for a long time
v1.9.1
i5 9600
16Gb Ram
Nvidia GTX1660
Windows 11

rossminney, 2024-03-04:
I'm also seeing this.
v1.9
i9 12900
64GB RAM
Windows 11 Pro
1TB 980 Pro nNVME
RTX3090.

When orca slicer is open the rest of the computer has extreamly slow response times (even just typing inside notepad++). Closing Orca slicer immediately resolves the issue.

radzish, 2024-03-04:
I do not think it is hardware/os related - it happens in different environmets, including mine (linux, a2000 mobile gpu, intel xeon processor, 64 RAM, 1T nvme drive).

I noticed though that this happens mostly when interacting with Orca in "Preview" mode. It slows down PC even if there is no interaction, but Preview mode is just open. Switching to Preview mode, makes everything responsive again.

Hygea, 2024-03-04:
Can confirm the same issue.
Win 10
Ryzen 7 1800x
64GB DDR4 RAM
1080ti

BloodSteyn, 2024-03-12:
Can confirm same issue, my laptop lags when Orca is open, even the mouse becomes laggy.

Windows 11 Pro
8 Core i7-11800H
32GB RAM
RTX 3060

CPU also runs over 90°c while showing almost no load in Task Manager. Once Orca is closed, temps go down below 80°c and the lagging goes away.

Dendrowen, 2024-03-13:
Same issue here. On both 1.9.1 and 2.0.0

Win 11 pro
Ryzen 9 5900HX
16GB  RAM

MistressPlague, 2024-03-16:
Same issue here. I suspect perhaps the log file being written to so much is the culprit. I could be wrong however.

Win 10 Pro
i7 9700K
64GB RAM
NVME SSD

vgdh, 2024-03-17:
sometimes have same issue
may be related https://github.com/SoftFever/OrcaSlicer/issues/997

Al3601, 2024-03-17:
Same Issue,

Ryzen 5600x
Nvidia 3080
32GB ram cl14 3200
3440x1440

JmsOG, 2024-03-20:
Same issue, Orca 1.9.1, got somme big delays when using any program on my computer. i dont see any spikes on usage on the task manager.

Intel I7 11700K
Nvidia 4060
32Gb Ram
W11 Pro 23H2 22631.3296

nesttle, 2024-03-21:
Same issue.

Orca 1.9.0
AMD 7800X3D
Nvidia 3070 8GB
32Gb Ram 
W10

wilkems, 2024-03-24:
Same Issue

Orca 2.0.0 Dev
Ryzen 7 3700x
NVidia 2070 Super
32GB Ram
W11 Pro 23H2 22631.3296

iEBPO, 2024-04-01:
Same problem. When previewing GPU 100%. It is absolutely impossible to do anything.

CPU i7 Core 10750
RTX 3060
16 RAM
W10 Home 22H2

mrhasbean, 2024-04-09:
Same issue
Orca 2.0.0
MacOS 13.4
4.2Ghz i7
48Gb RAM
Radeon Pro 8Gb
1Tb Samsung 980

Running at or above 100% while idle, no interaction.

pluciorx, 2024-04-22:
Same Issue here with:
Orca 2.0.0
XPS 15 9510
Win 11
i7-11800H @ 2.30GHz,
32GB Ram
RTX 4050
1Tb Samsung 980

Fariultra, 2024-04-26:
Same here, Windows 10 and RTX3070.
[image]

MikeMM98, 2024-05-08:
Same issue:

i7-4770 CPU @ 3.40GHz
NVIDIA GeForce GTX 1060 3GB
32GB RAM
Win10
Orca Slicer 2.0.0

Showing no load on CPU, GPU, or RAM. However, it is making my whole computer run slow even when I am not directly using the app...

Are there any stable versions out there?

JmsOG, 2024-05-08:
I don’t know if there’s really a link between the two but I was using Avast as an Antivirus and since i completely deleted it and now use Windows Defender i dont have that issue anymore. Just taught i’d share.

faith-healer, 2024-05-10:
@JmsOG I use ArchLinux and experience same issues - so no, it's not AV software related (at least not only them).

nikoramiro, 2024-05-11:
Suffering the same issue here. Seems like for some reason all the slicing is happening on the CPU.

Ryzen 9 3900X
RTX 3080
32GB RAM
Win11
Orca 2.0

akoolenbourke, 2024-05-13:
Same here.

win11
GTX 1060
32GB RAM
Orca 2

I was going to look to debug this but the build process for Orca is dodgy and noone seems to want to help.

emguepo, 2024-05-16:
Same problem

Orca Slicer 2.0.0
Using 1 monitor extra and Wacom Cintiq Pro 16

---------------------------------------------------------------------
Lenovo Legion Pro 7 16ARX8H
Win 11 23H2
AMD Ryzen 9 7945HX (Radeon 610M)
64GB RAM 5600 (5200MHz)
Nvidia RTX 4090 Laptop 16GB
1TB SSD Micron Crucial 7400MB/S Gen4 NVMe M.2
2TB SSD T500 Micron Crucial 7400MB/S Gen4 NVMe M.2
---------------------------------------------------------------------

pluciorx, 2024-05-16:
Could be duplicate of: #997

wilkems, 2024-05-16:
[image]

Opened up ProcMon, and this is all the activity associated with a tab switch. So perhaps it's read/write related?

MYexeto, 2024-05-16:
Ok, when switching tabs. But why whole computer lagging when you are not
switching tabs, orca slicer just remaining open while you are doing
something else in the other apps?

akoolenbourke, 2024-05-17:
> Ok, when switching tabs. But why whole computer lagging when you are not switching tabs, orca slicer just remaining open while you are doing something else in the other apps?
> […](#)

That  I am not sure of but I'd like to get to the bottom of it.

syyyn, 2024-05-19:
Just here to confirm issue for me was Avast Antivirus. Lag gone.

MYexeto, 2024-05-20:
> Just here to confirm issue for me was Avast Antivirus. Lag gone.

Also confirming. Lag gone after Avast uninstall.

### **Everyone please confirm if Avast was an issue for you and i will close this thread.**

mrhasbean, 2024-05-20:
I have the issue on a Mac and there was at least one other person with the issue on Linux. No Avast here.

agocska, 2024-05-20:
No avast here.

~ mobilról, útközben ~

Steve H ***@***.***> (időpont: 2024. máj. 20., H, 7:40) ezt
írta:

mattyg2301, 2024-05-20:
No Avast here either

mattyg2301, 2024-05-21:
@Al3601 

You legend, looks like that might have done the trick.

Made the changes yesterday and so far so good, nearly through my pc out the window :)

Will revert back if things change!

Thanks!

BloodSteyn, 2024-05-27:
@Al3601

I just tried your recommendations. It definitely works. I can actually scroll the settings without lag, and changing settings tabs is greatly sped up.

Thanks.

EDToaster, 2024-07-01:
For our English speaking friends, the settings above are

* Control Flow Guard (CFG)
* Data Execution Prevention (DEP)
* Validate exception chains (SEHOP)
* Validate heap integrity

[image]
[image]

Mechotronic, 2024-07-28:
I can't believe this is being presented as a solution. Don't turn that stuff off.

MYexeto, 2024-07-28:
Do you have a better solution for this?

The fun think that on my lenovo laptop I never had this issues.

Mechotronic, 2024-07-28:
Fix the software. Disabling important AV settings is not a solution.

EDToaster, 2024-07-28:
Obviously it's not the end solution, that's why the issue is still open. This is just a workaround for the meantime because the software was completely unusable.

MYexeto, 2024-07-28:
Yeap  it would be great solution, bit I'm not a developer to fix it)))

EDToaster, 2024-07-28:
The issue is present on PrusaSlicer as well. I've opened an issue there to track: https://github.com/prusa3d/PrusaSlicer/issues/13173

Alyx-Clark, 2024-08-03:
SOLUTION: Go into Windows settings - Display - Graphics - add OrcaSlicer from desktop apps and select high performance/dedicated graphics card instead of the integrated one.

This fixed all lagging for me in orca slicer. You may also have to right click desktop - show more options - nvidia control panel - 3D settings - manage 3d settings - click program settings tab - add orca slicer - select high performance - click apply at bottom

Windows 11
Nvidia Geforce RTX 4060 Laptop GPU

MYexeto, 2024-08-05:
> SOLUTION: Go into Windows settings - Display - Graphics - add OrcaSlicer from desktop apps and select high performance/dedicated graphics card instead of the integrated one.
>
> This fixed all lagging [...]

wellp, it didn't work for me )

nesttle, 2024-08-08:
Still laggy with last update

tealblu, 2024-09-07:
also experiencing lag on archlinux with Hyprland - the weird thing is the renderer is completely responsive but the rest of the UI is not (tabs, print settings, anything except the 3d renderer window)

EDToaster, 2024-09-10:
> the weird thing is the renderer is completely responsive but the rest of the UI is not (tabs, print settings, anything except the 3d renderer window)

this is my experience as well.

Blady144, 2024-09-28:
Enable the integrated graphics card in the device manager.

MYexeto, 2024-09-28:
I don't have integrated graphics :)

ofloan, 2024-11-04:
Slows down the system drastically. Havent had integrated graphics on for months. Deactivated in BIOS.
i9-14900HX / 4070 / W11 (laptop)

Kcdc3d, 2024-11-12:
I have dual RTX4090s on this workstation and this massive lag happens to me over time. If it's graphics related, it's gotta be a leak or poor performing/old openGL or what ever its used to render the viewport, maybe windows 11 but this seems too specific. This is the kind of stuff I used to deal with on major DCC programs, softimage and Maya way way back when and both times it was viewport inefficiency.., vram leaks, not really sure it's windows doing it, but possible a portion of whats used in Orcaslicer isn't supported by win11 as it was on win10. It's been out for a while now, so..

ofloan, 2024-11-20:
My MX Master 2S already has 125Hz (low) polling rate and my system has deactivated integrated graphics, so the problem happens regardless of these two settings.

vanowm, 2024-11-21:
@arnookie this has nothing to do with mouse setting, I can literally see how popup menu/windows are being painted on the screen when they opened, even though task manager doesn't show any high CPU usage by the Orca. My memory usage is also not the cause, it's within 50% and not being swapped to the page file.
In my case the majority of the slow down caused when one of the Orca slicer windows is opened in `Device` tab, and the slowdown dramatically enhanced if camera feed is playing.

badthumbs, 2024-11-26:
Run as Admin works for me....... 😉

vanowm, 2025-02-25:
Hate these useless bots, wasting everyone's time by sending useless emails to everyone subscribed and requiring "bump" the unresolved issues, which will send another useless email to everyone...

waynevanrooyen1971, 2025-03-03:
Same issue it annoying and no one seems to have a fix or answer from devs

[47 more replies]