OrcaSlicer/OrcaSlicer #3902: XY shrink compensation separately

Kokored7214, 2024-01-30:
**Is your feature request related to a problem? Please describe.**
Printing part with different angle comes with a different size. (Same part printing in same plate but different angle)

**Describe the solution you'd like**
We will need a filament setting that allow users to set the X and Y shrink compensation separately.
Right now OrcaSlicer can set a shrinkage percentage which automatically change the X and Y size.
But not all the printer got the same belt cal system as the bambu. Manualy cal might not be perfect that cause size error.

**Describe alternatives you've considered**
Using the current X-Y shrinkage % setting but it change both X and Y size. Also tried manually resizing but it cumbersome and not acting too well.

**Additional context**
This feature can improve the printing quality for most printer even the printer is bad.

Kokored7214, 2024-01-30:
But I think it can still be a feature like the z-offset one

LupusTheCanine, 2024-02-01:
Fix your printer. correctly set up printer must move as described in GCode, 10mm is 10mm not 10.05mm.

closed 2024-05-10 by github-actions[bot]